meshcore-dev/MeshCore #260: Sensecap T-1000e crashes regularly

Lucianpb, 2025-05-08:
Hello,

Bought a Sensecap T-1000e from Seeed, flashed it with Meshcore 1.5.1 and for a couple of days it was fine, however it seems to be crashing regularly.
Right now it is not reachable via BT and the led is not pulsing as it should, but it is continuously on.

Pressing the button or keeping it pressed for various periods of time does not reset it.
I can reset it by pressing the button while connecting the power cable to it, then it would come back online for a while more.

Any suggestions to help debug this further?

Thanks

recrof, 2025-05-08:
adding @fdlamotte - he might have clue what is wrong

fdlamotte, 2025-05-08:
@Lucianpb maybe you can switch to custom firmware ? 

you'll find it here https://github.com/fdlamotte/MeshCoreT1000e/releases

You shouldn't loose functionalities, but there are some tools in it that might help in debug ...

seagull9000, 2025-05-08:
This happened to me last week. 

Here's what I did:
- I did a hard reset using this technique:  https://www.youtube.com/shorts/9d7wQCJQAn0
- export settings using the app
- I used the full erase using the dfu file from the meshtastic flasher web site
- You will need to manually enter DFU by doing the hard reset twice in quick succession 
- Save the full erase to the T1000E
- Reflash with meshcore using the meshcore web flasher
- Import the settings you saved using the app

Now the T1000E is stable again.    I think it must have been a file system corruption.

chrstnwhlrt, 2025-05-12:
@seagull9000 Thanks for the instructions. Can you add the link to the full erase firmware please? I currently can't reset my device..

recrof, 2025-05-12:
@chrstnwhlrt https://flasher.meshtastic.org/uf2/nrf_erase_sd7_3.uf2

chrstnwhlrt, 2025-05-13:
Okay the issue is not (only) related to rebooting, just keep the device laying around, connect, disconnect using bluetooth and after some time it is not reachable anymore. Only way to reconnect is to enter and leave the DFU mode by quickly reconnecting the charger while pressing the button

fdlamotte, 2025-05-13:
Also another thing you can check (because it happens).

Can you confirm that there is no other phone that "steals" the connection ?

oltaco, 2025-12-02:
This issue seems like it might be a duplicate of https://github.com/meshcore-dev/MeshCore/issues/1067 where all the traction is now taking place.

Maybe we can close this one?

closed 2025-12-02 by liamcottle