meshcore-dev/MeshCore #275: Option to control public view of position for GPS-enabled devices

seagull9000, 2025-05-12:
From v1.12.0, we can use the device GPS to get the exact position, and we can allow other selected devices to see that exact position.

However, an Advert will also make the precise position available to all that receives the advert, which might not be the desire effect for some.

Suggest the following potential solutions:

1.  User switch in the App to not send the GPS location in an Advert
2. User-selectable precision in teh App to send GPS location in Advert

If we do the second option, it would be good to have a precision ring around the exact position that the user can see on their App map as a comfort that the selected precision is in effect for non-authorised devices.

446564, 2025-06-20:
we can also use the MT way and just allow imprecise, knocking digits off the coordinates as we decrease granularity for adverts only, so on sending the advert it would change the coordinates first. then contacts with location permission can still get your real location but adverts will be imprecise.

mattsains, 2025-06-20:
> we can also use the MT way and just allow imprecise, knocking digits off the coordinates as we decrease granularity for adverts only, so on sending the advert it would change the coordinates first. [...]

I think I prefer this way - I do want to share my rough location, so that people know where there is coverage, but I don't want someone knocking on my door to talk about MeshCore

However, if that's too hard to implement, I think 1 is okay, I think it should be off by default (do not share location by default)

ripplebiz, 2025-06-24:
Related to this, there is now a new preference in companion firmware, for whether you allow your location to be included in your adverts. Is in dev branch. Liam is adding a new toggle in the app for turning this on/off.
So, this is independent of telemetry request/response. You can separately control location permission for those.