meshcore-dev/MeshCore #276: linting rules

446564, 2025-05-12:
Should we decide on some formatting roles for linters?

I have format on save which I can of course disable but was wondering if we wanted to settle on some rules for formatting.

446564, 2025-05-16:
for now I have just turn off formatting to avoid rewriting the whole file when I change one line.

Hakira-Shymuy, 2025-05-16:
i second this

actually im part of a big community in portugal, and we are looking to adopt a network and protocol for our safe transmissions and emergency support, im talking probably around 500 people on active use and close to 3k as curious users, and it did catch our attention that meshcore did solve some stuff that meshtastic did not

at the same time we are reluctant due to the fact that the repository is a bit greeny
Would be nice to see the repo under one organization rather than a user, and some contribution guidelines, CI/CD i see that you guys are using for release versions, git convetional commits (https://www.conventionalcommits.org/en/v1.0.0/)

definitely would create more trust

Cheers

Hakira-Shymuy, 2025-05-16:
clang-tidy and cppcheck might worth to take a look at

446564, 2025-06-16:
looks like we have a `.clang-format` now, closing

446564, 2025-06-19:
I think we should evaluate the currently included clang format configuration.

It seems to indent with spaces instead of tabs and breaks structures in a way that seems odd or at least not in line with the current codebase. As it stands if you save an existing file using this configuration a whole lot of formatting changes. 

After that the entire project should be linted and formatted at once so we aren't changing entire files when making small changes.

Hakira-Shymuy, 2025-06-19:
> I think we should evaluate the currently included clang format configuration.
>
> It seems to indent with spaces instead of tabs and breaks structures in a way that seems odd or at least not in line [...]

The issue with the formatter Rules is that 

`UseTab:          Never`

So it will use spaces for the indentation
either use

`UseTab: ForIndentation`
or
`UseTab: Always`

in line 83

closed 2025-09-16 by liamcottle