meshcore-dev/MeshCore #277: Missing messages in Public Channel

seagull9000, 2025-05-12:
Strange one with a gap in message history on the Public channel.

App v1.12.0 Android Pixel 7
T1000e latest dev v1.6.0

Last message before gap: 6 May 20:24
First message after gap: 13 May 05:13

Perhaps a couple of dozen messages went AWOL between these times.

History:
T1000E has been rebooted/updated several times during development in the last day.
I had sideloaded App v1.12 before I realised the App Store has updated, so I uninstalled the App and reinstalled from Playstore.

recrof, 2025-05-12:
smells like database corruption

seagull9000, 2025-05-17:
Adding a couple of further issues:

While the App does stay connected for longer, after a while it fails to notify when a new message is received or a new node is found.

Sometimes when foregrounding the App on the phone it will come up with "Failed to sync" and it will either miss messages (suspected), or drop contacts which must then be readded from the discover contacts list.

The App v1.12.0 is the one downloaded from the Android Playstore.

oltaco, 2025-12-02:
Just checking over the issues and noticed this one hasn't been updated in over 6 months, and there's been a lot of changes since then.
Maybe it can be closed?