meshcore-dev/MeshCore #283: Add support for INA3221 current sensor to the Repeater firmware

ngavars, 2025-05-13:
**My Use Case**

A solar repeater node that is using non-lithium battery (Lead Acid, LTO, Na+) often would use a DC-DC buck/boost converter to power the node. This means that the node would not be able to use the built-in ADC to measure battery voltage. Hence the requirement for current/voltage sensor like INA3221.
INA3221 is particularly useful, because it has 3 channels and would be able to measure also solar panel output and, possibly, other voltages.

It would be great if it would be possible to current/voltage/power values from all 3 INA3221 channels somewhere in telemetry section of the app (or via meshcore-cli).

ngavars, 2025-05-13:
I have added support for INA3221 to the Promicro variant. I'll be creating a PR shortly.

oltaco, 2025-12-02:
This can be closed, can you do the honours @liamcottle