meshcore-dev/MeshCore #299: Display of message on companion node on wrong channel

seagull9000, 2025-05-17:
The companion radio displays message received on Public, when it was sent to a private channel.

It shows 

(0) Public
_message_

When the App is connected to the companion radio, the messages comes up on the private channel correctly.

The sender is a T114  connected via USB to a server with meshcore-cli sending messages to the private channel.

The receiver is Heltec Tracker running v1.6.1 (was happening under v1.6.0 also).

446564, 2025-05-18:
Like like "Public" is hardcoded https://github.com/ripplebiz/MeshCore/blob/9048142f63db60aacc5e80c7b53f8ab96c1a9293/examples/companion_radio/main.cpp#L664

seagull9000, 2025-05-18:
That would do it!  Thanks, I spent some time looking at the code and completely missed that.

446564, 2025-05-18:
There should be a `name` for the `channel` being passed if you want to test it.

liamcottle, 2025-06-03:
Can be closed out, implemented in #356

seagull9000, 2025-06-03:
Great!  Thanks all.

closed 2025-06-03 by seagull9000