meshcore-dev/MeshCore #309: GPS update interval

446564, 2025-05-19:
I was working on the nano g2 ultra and thought that 1s is too often to poll the GPS for position.

We should probably think of a good interval after the fist fix to save some CPU cycles aka battery.

446564, 2025-05-20:
Also maybe the gps should be put to sleep or standby between polling?

prusnak, 2025-08-09:
This applies to other variants such as T1000-E 

I think it would be best to poll every 1 second if `location->isValid() == false` and poll every 60 seconds if `location->isValid() == true` ?

446564, 2025-08-09:
Ya this was meant for all variants.

I like that though, reduce polling once we have a fix.

We should also try to get the loop out of variants and into the environment sensors package so it can be standardized.

prusnak, 2025-08-09:
> We should also try to get the loop out of variants and into the environment sensors package so it can be standardized.

Exactly 👍

oltaco, 2025-12-02:
I believe gps_interval was introduced in https://github.com/meshcore-dev/MeshCore/commit/6ed8e9d5142715b61a22fed6dc8627fb3b3476cc so we can probably close this now?

closed 2026-10-12 by github-actions[bot]