meshcore-dev/MeshCore #319: LilyGo T3S3 LR1121 support

vlszabolcs, 2025-05-21:
Is LilyGo T3S3 LR1121 board support planned?

recrof, 2025-05-21:
I have experimental builds for this board if you're interested. the main problem is dual tx power required for different bands. until we figure this out, it won't be in the repo.

vlszabolcs, 2025-05-21:
Is the main problem hardware or regulation specific? Could you explain exactly what it is about!

vlszabolcs, 2025-05-21:
Is the main problem hardware or regulation specific?

recrof, 2025-05-21:
> Is the main problem hardware or regulation specific?

the main problem is, that we don't have logic to support multiple tx power limits. currently we only have one setting that is universal. if you switch from 868 to 2.4Ghz, it needs to change from 22dBm to 20dBm. in my experimental firmware, this stays at 22, which could be potentionaly problematic.

recrof, 2025-05-21:
[LilyGo_T3S3_lr1121-experimental-1.6.2-dev.zip](https://github.com/user-attachments/files/20365564/LilyGo_T3S3_lr1121-experimental-1.6.2-dev.zip) here are all most used roles for you to try.

liamcottle, 2025-09-16:
Closing for now. Haven't seen any PRs to add support.

closed 2025-09-16 by liamcottle