meshcore-dev/MeshCore #320: Firmware Updates over Radio Link

mikecarper, 2025-05-21:
Fastest possible LoRa speed is 37.50 kbps using SF of 6, bandwidth of 500 kHz, and a 4/5 Coding Rate.
The SX126x supports FSK modulation with data rates up to 300 kbps.
2.4ghz LoRa should be able to go fast as well, but you can update the firmware over wifi in theory then. 

Targeting updating over LoRa with the bandwidth at 250 kHz is the best way forward as it'd have the widest support in theory.

mikecarper, 2025-05-21:
https://components.espressif.com/components/espressif/esp_delta_ota  
https://github.com/zephyrproject-rtos/zephyr/issues/78529 https://github.com/Scionova/zephyr-delta-updates  
https://github.com/Thisora/wolfBoot-rp2040

mikecarper, 2025-07-26:
Target the station G2 using LoRa SF7/250 kHz
 https://github.com/espressif/idf-extra-components/tree/master/esp_delta_ota/examples/https_delta_ota

Split delta patch into chunks with ordered filenames

Send data and save into 
/spiffs/0001.d, /spiffs/0002.d /spiffs/0003.d etc

Combine into
/spiffs/update.patch

esp_delta_ota_apply("/spiffs/update.patch"); 

reboot

Should take around 5 minutes to send a 100kb file to the device over lora. Roughly 500 packets of data with a payload of 200 bytes. Ack back should be a hash of the data chunk sent.

liamcottle, 2025-08-31:
This would be interesting to try out, and is something I'd like for updating very distance nodes, however this is unlikely something the core team is going to work on. If you decide to give it a go, and get it working, feel free to follow up :)