meshcore-dev/MeshCore #328: Feature Request - GPS on Heltec v3

mrlmuk, 2025-05-23:
Now that GPS is available on the Heltec Tracker would it be a lot of work to also enable it on a Heltec v3?

Looking at the variants/.../platformio.ini they both appear to be based around the same chip and board design so perhaps most of the work is already done. They differ in the display and the GPIO pins would need to be set correctly for GPS. I will have a go at a custom build / variant.

mrlmuk, 2025-05-25:
Found this will read more ...
http://github.com/ripplebiz/MeshCore/pull/91
Basic gps support for Heltec_v3 and T1000e

oltaco, 2025-06-21:
I can implement this, are there standard pins to use for the GPS on heltec?

oltaco, 2025-06-21:
Submitted a PR for GPS support on Heltec V3. Tested with Ublox Neo-6M. Connect RX on GPS module to pin 45 and TX to pin 46.

Note that while you can enable and disable the GPS there is no way to actually power down the GPS without using a mosfet and adjusting PIN_GPS_EN.

mrlmuk, 2025-06-21:
Thanks for having a look at this. 

Those were the pins which seem convenient and others have used such but I imagine this would always be a custom firmware someone would configure and build anyway. I'm aware of the battery drain issues and inability to turn it off I can figure out the hardware side and add a switch or mosfet easily as shown in these links.

https://www.thingiverse.com/thing:6967414. While this case is nice it has some frustrating design quirks which waste space.
https://www.reddit.com/r/meshtastic/comments/1i2obfg/has_anyone_added_a_gps_to_their_heltec_v3_i_have/
https://meshtastic.org/docs/hardware/devices/heltec-automation/lora32/peripherals/ 

I will see if I can make sense of what you have done to test it later to try and give some helpful feedback.

DaBosh, 2025-06-21:
i have one with this pinout, can offer in app to select gpio pins for tx nd rx

[image]

oltaco, 2025-06-21:
> i have one with this pinout, can offer in app to select gpio pins for tx nd rx
> 

I don't know of any plans to support changing the pins in app.

mrlmuk, 2025-06-25:
I have managed to build and install the firmware by editing the 1.7.0 files like you did in your commit. Have not connected the GPS  because I don't want to solder it yet and only have one quick pogo connector. Expected to see something in the software but cannot.

What does this do -D PIN_GPS_EN=-1 ?
Is this the pin which would control turning the GPS off and should be changed to a real pin number if the mosfet was installed? I assume -1 isn't a real pin and its some kind of placeholder?

oltaco, 2025-06-25:
Yes, if you are using a MOSFET to switch power to the GPS then you should set PIN_GPS_EN to the pin controlling the MOSFET.

Secondly you will need to make sure that the GPS has power when the node starts up, because it checks for GPS at power on, and if it is not detected then the options for controlling it will not be available in the app.

I guess you will want to make sure that the pin controlling the MOSFET is pulled high by default, I'm not sure if the GPS code does this but I will put it on my list to check.

liamcottle, 2025-08-31:
Closing this one out as OP says you need to modify the hardware, and there are other boards by the same manufacturer that already have GPS built in.

closed 2025-08-31 by liamcottle