meshcore-dev/MeshCore #333: Button class with callbacks proposal

seagull9000, 2025-05-26:
This is a proposal to add a button class to the UI to aid development of new features and to maintain consistency across devices.

It also help map out a button strategy / user experience.

The class should also support multiple buttons (e.g T-beam) and possibly menu control for devices with a acreen.

This is a starter for feedback on the idea.

So far I can see the following can be used for the user button:

- Long-Long (LL) press (> 4s): Shutdown (or deep sleep where shutdown not supported)  - available for all firmwares
- Long (L1) press x1 (> 1.5s < 3s): 
    - If a message is displayed on screen, dismiss it (companion firmware)
    - If a menu item is displayed, sends OK (all firmwares)
- Short press x1 (S1) (<= 1s)
    - if screen is off, turn it on (all firmwares)
    - if device sleeping, wake it up (all firmwares)
    - If screen is on, and message is displayed, advance to next (companion firmware)
    - If menu item is displayed move to next menu item
- Short press x2 (S2) - double press
    - Send flood advert (all firmwares)
    - if menu item is displayed, toggle setting (all firmwares), then L1 to OK
- Short press x3 (S3) - triple press
    - Toggle repeat on/off (repeater/room server firmwares)

These are just suggestions and other combinations could potentially be used (e.g. SL (short-long).

A call back function can be registered for each type of press (LL, L1, S1, S2, S3, etc).

hank, 2025-05-27:
Perhaps we could put a new set of virtuals in mesh::MainBoard to implement these, then stub them out in ESPBoard and the other child classes with either empty implementations or some standard behavior. That way we can just override them in the board classes. For the TBeam Supreme at least, this is implemented with XPowersLib at least in part:

PMU.setPowerKeyPressOffTime(XPOWERS_POWEROFF_4S);

The other parts seem to be handled by UITask::buttonHandler. Perhaps the buttonHandler could call the callbacks.

The only thing I would worry about is the consistency of the UI across devices and keeping that unified through changes.

seagull9000, 2025-05-29:
See jquatier's PR for potential implementation ... https://github.com/ripplebiz/MeshCore/pull/343

seagull9000, 2025-05-31:
Will close this as the above PR has implimented a button class.

closed 2025-05-31 by seagull9000