meshcore-dev/MeshCore #345: T1000-E Goes to sleep or deaf

MichaelHolroyd, 2025-05-29:
When I disconnect from my T1000-E after about an hour or so it seams to go deaf or to sleep.
If I connect in the morning and then disconnect and go to work when I come home and reconnect the device has stopped receiving router adverts.
The device also doesn’t collect any messages on the public channel during this time, and when it reconnects to the app it doesn’t update.

[image]

mjvolante, 2025-05-31:
I have also found since fw1.6.2 that when the battery goes dead, once recharged, my local Repeaters and RS display the default date of 2024. This is corrected when they advert.

[image]

mjvolante, 2025-06-01:
Like you, also finding T1000e going deaf when not connected to Phone, or at least not updating dates and times of adverts that occured whilst not connected. At around 1pm yesterday, got home, disconnected from T1000, and connected to home node. Just connected again and as image shows, no date or time update on nearby nodes. MJV Truro home node image for comparison. 

NOTE: T1000E RUNNING 1.6.0 AS I ROLLED BACK FROM 1.6.2 TO SEE IF ISSUES WENT AWAY.

[image]

[image]

mjvolante, 2025-06-01:
@fdlamotte Can you help with this?

fdlamotte, 2025-06-01:
Ok, I've load a firmware from the flasher, let's see what happens ;)

fdlamotte, 2025-06-01:
We really lack diagnosis tools for this kind of issues ...

Given the latest problems I had, I'd easily suspect some problem with storage. There are weird things that happen there and that's why we sometimes advise to erase the device (and it often works) ...

I'm thinking about a firmware that would be like the disk utilities of the computers in the 80's, to get an insight about disk usage, list files, manage them and verify their integrity ...

From one device to another, the only thing that changes is what is in the device memory ...

liamcottle, 2025-06-01:
> @fdlamotte I'm thinking about a firmware that would be like the disk utilities of the computers in the 80's, to get an insight about disk usage, list files, manage them and verify their integrity ...

I built a very basic file browser for nRF52 devices which lets you connect from a web page over BLE. This was about a year ago when I was needing to pull files from RNode Firmware. Maybe it could be useful :)

https://github.com/liamcottle/nrf52-file-manager

MichaelHolroyd, 2025-06-01:
I have just seen this happen in real time
My T1000went flat, I booted it back up, and reconnected to my app.
Just as it was refreshing the contacts etc all the dates and times went backwards.
Could this be to do with the clock in the T1000 not being in sync until it’s been connected?
Just a thought based on what I have seen.

fdlamotte, 2025-06-02:
From what you describe, I'd say the app had a more up to date contact list than the t1000.

I'm quite sure that is because the t1000 fails at writing contacts to its internal memory.

You should probably back up your keys and erase the internal memory.

recrof, 2025-06-02:
@fdlamotte I believe so. 1.5.1 was last stable firmware for nrf52 devices. no idea what changed that would cause this.

seagull9000, 2025-06-03:
There has been a number of suggested user actions on discord in the event of this (and other nRF52) device experiences this issue:.

For users, it can be a bit of a discovery finding all the information, so I will attempt to summarise them here people can link to this comment if helping someone out.

This is a rough starter so please add comments if I have missed anything and I wil update this.

In summary, I believe there are 3 visible issues:
1. Bluetooth connection fails and won't reconnect with the App.
   - Suggested user action:
      i.  On Android, force-close the App by swiping it away from the recent Apps list.  [Does iOS have this issue?]
2. Corrupted FS, times wrong on node list, missing messages or missing nodes, going to sleep, only waking when holding the button when plugging in
    - Suggested user actions:
      i. Hold the button while plugging into power to restart the device.
      ii. Use the App to back up everything,
      iii. Erase the flash memory on the device.  Get it into DFU mode, download the appropriate firmware erase uf2 for the device [link to somewhere].  Remember you need to open the device in a terminal (e.g. minicom) for the erase to complete ```minicom -D /dev/ttyACM0 -b 115200``` will do it in linux
     iv. Flash the last known good meshcore firmware (v1.5.1) using the flasher https://flasher.meshcore.dev/
    v.  Restore settings and data using the App.
3. I can put the device into DFU mode, but erase and flash doesn't work.  Device does not boot into meshcore at all.
If this has happened, it could be the bootloader has corrupted.  This is usually seen using a teminal (e.g. minicom) and the device is stuck on "Formatting ...".
    - Suggested user actions:
    i.  Follow the instructions here https://wiki.seeedstudio.com/sensecap_t1000_e/#flash-the-bootloader
   ii.  Follow the instructions for erasing the flash memory above and flashing meshcore in issue 2.

oltaco, 2025-06-04:
> Just noticed, that while the device was in this "crashed state", messages hadn't arrived, so I must be having both the Bluetooth issue and some storage corruption issue. This time after reconnecting, my contacts list was empty too.

What device is this? You mentioned earlier that you did some testing with Faketec but it was OK after 6 hours, is that still the case?

recrof, 2025-06-30:
> Anyone have a update on this at all?

we are trying to reproduce the issue without any luck. if you have multiple affected devices, you might want to either donate or tell us hardware revision of the device.

rookster1984, 2025-07-05:
Hi I am having this error again, I came from meshtastic again and tried it on the newest version of meshcore and after a hour or so with no phone connected to it I can't message it from another device or get it's location.

seagull9000, 2025-07-05:
The 1.7.1 and 1.7.2 versions have an issue with DM.

Are you able to get a message through with flood routing?

rookster1984, 2025-07-05:
[image] do you mean like this

rookster1984, 2025-07-05:
But if I connect to it via Bluetooth and send a message out from it, it starts working again.

rookster1984, 2025-07-05:
Thanks Will try and let you know

rookster1984, 2025-07-06:
Thanks, I tried overnight and it's still the same, it worked fine on mt. I wiped it and installed how you said

lincomatic, 2025-08-15:
> [@rookster1984](https://github.com/rookster1984) please try to flash this firmware via https://flasher.meshcore.dev/ - last option: Custom firmware. (don't unzip it, upload it as is)
> 
> [t1000e_companion_radio_ble_max_airtime_fix.zip](https://github.com/user-attachments/files/21081101/t1000e_companion_radio_ble_max_airtime_fix.zip)

What magic sauce is in this build? Is the change in the dev repo? With 1.7.4 my T100E was freezing up (LED not blinking, not responsive, couldn't power cycle holding the button) within 1 hr. It couldn only be awakened by applying power via the cable (it would never lock up w/ external power). With 1.7.3, it would last a few hours. This one is working overnight.

fdlamotte, 2025-08-15:
not sure about 1.7.3 but 1.7.4 should already have the fix ... 

maybe a corrupted fs ...

fdlamotte, 2025-08-15:
How many contacts ?

There is a big fs corruption bug on all nrf variants ... its currently worked on (but I'm not part of that work so don't have much details)

DevertNet, 2025-08-27:
I have the same problem with 1.7.4:
- Freeze / Sleep after some time
- Not connecting via bluetooth
- Not responding to the button
- When connecting with the cable/power it start again and it works

What i tried so far:
- Erase with the uf2 file from the flasher page
- flash 1.7.4 again

As next step i will try the custom firmware: 
https://github.com/meshcore-dev/MeshCore/issues/345#issuecomment-3040126203

IoTThinks, 2025-09-13:
@recrof I will open another issue to ask.
Seems related to UF2 vs. DFU bootloader.

orsib78, 2025-09-29:
Similar problem for me. I cannot initiate DFU mode using the web flasher and it is also affecting my t-echo as well. When I click on the DFU button my t1000 simply reboots.

I also tried adafruit-nrf tool from the command line. But no luck. It seemingly connects to the device but then it fails due to lack of answer from the t1000. So the only option for me is to drop an .uf2 file to the device.

> > Using the Web Flasher (neither DFU nor Flash) does not work. For me, it only works with the UF2 files and manual DFU mode
> 
> Me too. Have not been able to flash my T1000-e using Web flasher. Only manual DFU and drop in uf2 file.
> 
> Clicked DFU mode active
> 
> <img alt="Image" width="1917" height="960" src="https://private-user-images.githubusercontent.com/29994971/488811275-486fc7c5-dc7c-4835-b4f6-c7613d64f2a6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxNDg5ODAsIm5iZiI6MTc1OTE0ODY4MCwicGF0aCI6Ii8yOTk5NDk3MS80ODg4MTEyNzUtNDg2ZmM3YzUtZGM3Yy00ODM1LWI0ZjYtYzc2MTNkNjRmMmE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDEyMjQ0MFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTAzZjNjMjlhNDMxMjViZjcxZTliYjE1NTkxY2QwMjVkYjA3NDA5MjkwZDBkODc4OTZhYzA3NjA3NzViNmYyZTgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.LO0R8wTa7aZBTAXyQXSDw-daKTvfjRk_U4Cq-NERWCo">
> Then Erase Flash / Flash and hit error <img alt="Image" width="1781" height="621" src="https://private-user-images.githubusercontent.com/29994971/488811016-a565bcaa-b7e4-43e3-8775-a57546cc2761.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.

> > Using the Web Flasher (neither DFU nor Flash) does not work. For me, it only works with the UF2 files and manual DFU mode
> 
> Me too. Have not been able to flash my T1000-e using Web flasher. Only manual DFU and drop in uf2 file.
> 
> Clicked DFU mode active
> 
> <img alt="Image" width="1917" height="960" src="https://private-user-images.githubusercontent.com/29994971/488811275-486fc7c5-dc7c-4835-b4f6-c7613d64f2a6.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxNDkwMjQsIm5iZiI6MTc1OTE0ODcyNCwicGF0aCI6Ii8yOTk5NDk3MS80ODg4MTEyNzUtNDg2ZmM3YzUtZGM3Yy00ODM1LWI0ZjYtYzc2MTNkNjRmMmE2LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDEyMjUyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWUzMzk2ZGY0ZDcxOTE5NzZkMTQ0MjgzNWIxMDBkM2RkN2ZlMWMzZTE1NmFlOGE0Y2RmNGU1N2E0YTcwMzlkODQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.6BWEt8GFihoZbXN3wR3HZN5DgPFUBp0aer4R17syk6E">
> Then Erase Flash / Flash and hit error <img alt="Image" width="1781" height="621" src="https://private-user-images.githubusercontent.com/29994971/488811016-a565bcaa-b7e4-43e3-8775-a57546cc2761.png?jwt=eyJ0eXAiOiJKV1QiLCJhbGciOiJIUzI1NiJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTkxNDkwMjQsIm5iZiI6MTc1OTE0ODcyNCwicGF0aCI6Ii8yOTk5NDk3MS80ODg4MTEwMTYtYTU2NWJjYWEtYjdlNC00M2UzLTg3NzUtYTU3NTQ2Y2MyNzYxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTA5MjklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwOTI5VDEyMjUyNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWFiMzU5NGIyMDg4YTczZDRhZTBjMzk2NmIzOTBhMzkzMzk0NGExOGRkNTg2NWE2NjdjNjRkYjY2NzQwNDA4YmEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.C0i4oEcgnv__yyW7tfSfAhg1E7hETUKJ4ifKPsgDvG0">

closed 2025-10-17 by fdlamotte