meshcore-dev/MeshCore #363: Version 7.0 on T-Deck plus

pdm0393, 2025-06-03:
When name of node is on my homescreen i can see that there is a direct message from that node. If node is not on my homescreen i hear the sound that there is message but dont see it anywhere. is  that bug in software?

mattsains, 2025-06-12:
I think you need to have received an advert from a node in order to read direct messages from it. This is because the message is encrypted with the node’s private key, so you need the public key (contained in the advert)

pdm0393, 2025-06-13:
yes but on companion app names can be added automaticly and then you can send messages directly

peteforsyth, 2025-07-15:
I'm not sure how to reflash the T-Deck. It doesn't show up under USB devices when using the web flasher. I don't need to preserve settings or contacts...is there a way to flash it without the web flasher? I'm on version 6.8, looking forward to updating soon.

peteforsyth, 2025-07-15:
> [@peteforsyth](https://github.com/peteforsyth) are you holding the trackball button while powering it up to get into usb flash mode? if yes, did you try different usb cable or different computer? if the device is not visible for web flasher, it won't be for any other flashing program.

Thank you for this. I had seen something about pressing the track ball, but your message got me to redouble my efforts. I had two stumbling blocks:

1.  I found I had to press the track ball _after_ the boot sequence starts, not _before_. Then it would start up with a flashing orange border, which I understood to indicate the USB flash mode.
2. Chrome on Linux did not work this time, though it has on some devices in the past. Rebooting to Windows, once I had the device properly in USB Flash mode, did the trick.

So, I'm all set! Many thanks for your help.

However, I'm wondering. It seems to me that there is not a comprehensive central "how to flash various devices with Meshcore" resource. Since I'm dealing with several different devices, I could try to take notes and collect this info to build something. Would that be helpful to the project? If so, where's a good place to work on this?

liamcottle, 2025-08-31:
From TDeck firmware v7.7+, only contacts you've added to your homescreen can DM you now. So this shouldn't happen anymore.