meshcore-dev/MeshCore #370: Mescore not working on T5_EPAPER

renniehill, 2025-06-04:
Tried 3 times, without success, (definition of madness?) to flash MeshCore (V6 once, V7 twice) onto T5-EPAPER (sticker code 201-220052) (ESP32-S3/LORA on module) using second device option on MeshCore Flasher list.  Device not responding but screen now intermittently flashing brightly. Screen flashing stops when RST is pressed but resumes after about 10 seconds.  During programming. serial comms connected fine, then writing to Flash appeared to proceed and complete correctly. Unfortunately, not the sort of flash operation intended...

mr-tbot, 2025-06-07:
Same issue here - just received a T5-Epaper S3 Pro from Amazon - it appears this new version with the GPS chip has issues with Meshcore - can confirm same behavior - screen flashes intermittently after what appears to be a successful flash.

Defeator2020, 2025-07-04:
Also very interested in support/an avenue to modifying the existing firmware to work with the GPS variant of the T5

closed 2026-10-12 by github-actions[bot]