meshcore-dev/MeshCore #388: Turning off Meshcore on Heltec V3 not working

alrik11es, 2025-06-11:
As the title implies, I can't turn off my Heltec V3 running Meshcore. I've tried multiple versions. In Meshtastic, for example, it turns off by holding the function button for 3–5 seconds, but that doesn't seem to work here.

tarastella, 2025-06-12:
Let me say that it's not urgent.
I have a Muzi Works R1 with an external antenna, based on WisBlock RAK4631.
In Meshtastic, the application allows for the shutdown of the device. Pressing the button for 5 seconds will also shut it down.
It's not healthy for the device, but I'm just waiting for the battery to drain and that's it. However, if you ever consider fixing this for the Heltec V3, you might also want to take a look at these types of external devices. Thanks a million.

ErueWoad, 2025-07-19:
I would also like the ability to turn off my Heltec V3's

RodMyers, 2025-07-27:
adding my voice for this necessary feature

4np, 2025-08-15:
AFAIK you also need to power off a device to switch antennas as removing an antenna from a device actively using Lora may damage the hardware. My T114 (companion) and RAK4361 (companion & repeating room server) devices can only power down by pulling the battery plug which is a bit cumbersome.

For repeaters / room servers, perhaps a command line command can be introduced to power down the radios for a given number of seconds (`lora off 120`, `ble off 60`)? Maybe with a led blinking slowly and more rapidly nearing the end of the timer?

liamcottle, 2025-08-31:
These devices don't actually support power off. They're more of a "sleep" or "hibernation" state, where they still use power. The new UI that is in the dev branch, and should be available in v1.8.0 firmware will have the ability to hibernate the device. I'll close this out since that's been implemented already.

closed 2025-08-31 by liamcottle

Jamira40, 2025-09-01:
Flashed Heltec v3 with latest 1.8.0 and hibernation looks like its not working correctly. Device is still connected and via BT and picking packets.

olanwe, 2025-11-09:
> > To turn off my heltec, I need to hold down the top button and then press the bottom one.
> 
> This seems to do something. I will give it a try. Is this a build in Feature of the Heltec?

Doesn't that key combination simply put it into boot mode (reset while prog-button pressed)? So it's not really off, it is just waiting for commands on the serial line.