meshcore-dev/MeshCore #392: refactor sensors

446564, 2025-06-14:
## Situation

Right now environmental and other sensors are implemented on a per variant basis, some more tightly coupled than others. This makes it tricky to add sensors in general or to expand support for an existing variant. Requiring build flags, among other things, means users need to custom build firmware.

## Proposal

I propose that we move all sensors out of variants and into the sensors library, and change to using a scanner to detect sensors and enable their use when found.

Consideration will need to be given to potential address conflicts, when an address may conflict with another sensor will need to figure out a way to decide which of the two or more it may be. In the event that more than one are present with the same address, I propose that the first to be found is kept and subsequent sensors are skipped and log an error somewhere.

## User Story

- As a user I should be able to attach a supported sensor to a supported device variant running firmware from the web flasher or github and have sensor data work out of the box.
- As a user I should be able to add support for a new sensor by looking at the sensors library and adding a new definition for the address, telemetry etc and it should then work on any variant using that sensor.

## Variants to be refactored

*Some may not have sensor support yet so at this time we could consider adding it.*

- [ ] generic-e22
- [ ] generic_espnow
- [ ] heltec_tracker
- [ ] heltec_v2
- [ ] heltec_v3
- [ ] lilygo_t3s3
- [ ] lilygo_tbeam
- [ ] lilygo_tbeam_SX1262
- [ ] lilygo_tbeam_supreme_SX1262
- [ ] lilygo_tlora_v2_1
- [ ] nano_g2_ultra
- [ ] picow
- [ ] promicro
- [ ] rak3x72
- [ ] rak4631
- [ ] station_g2
- [ ] t1000-e
- [ ] t114
- [ ] techo
- [ ] thinknode_m1
- [ ] wio-e5-dev
- [ ] wio-e5-mini
- [ ] xiao_c3
- [ ] xiao_nrf52
- [ ] xiao_s3_wio

446564, 2025-06-14:
I wasn't sure how we sold approach on regards to PRs, obviously one big PR isn't great. I was thinking at first to implement the scanner and add support for all the current sensors, then migrate each variant one at a time.

fschrempf, 2025-09-08:
Great suggestions already in the posts above. Just to sum up and add my own 2 cents:

* we have features such as sensors (but also could be other optional peripherals like displays, buttons, etc.) that depend on hardware setup
* handling each potential hardware setup as a hardcoded variant is not feasible

Possible solutions:

1. built-in hardware support with runtime configuration by user
2. built-in hardware support with autodetection
3. user can use web tool to create individual firmware for their setup with peripheral configuration

1 and 2 are especially problematic for devices with small flash size. All three could also be combined in some way.

Just for reference, there are other projects and frameworks with similar solutions that come to my mind:

* The [Zephyr project](https://zephyrproject.org/) uses devicetrees (derived from Linux kernel) to describe the hardware setup and overlays to describe extensions. This makes the firmware as modular as the hardware.
* [ESPHome](https://esphome.io/) uses YAML files to describe the hardware setup and make it configurable for the user without code interaction.

closed 2026-10-12 by github-actions[bot]