meshcore-dev/MeshCore #398: support other battery chemisrties

446564, 2025-06-16:
## Situation

Currently the firmware assumes a battery to be lithium polymer with a working range of 3-4.2v.

Other chemistries that may be more suitable to remote nodes operate at different voltages. We could just use an INA board the get the voltage and/or current and expose it on another telemetry channel, however this is a bit clunky if your primary battery is not lithium as the UI in companions or the various client apps will show 0v.

Another related issue is that battery percentage is derived from the assumption that we have between 3 and 4.2v. Some of this may be happening in the companion apps (mobile/web), but the firmware can only return the millivolts for the battery which does not reflect the actual remaining capacity for any chemistry beyond lithium.

## Proposal

There are at least two important things to change, one is the addition of battery percentage as well as voltage. Another being basic battery curves. Even as crude as lipo being 1.2v range, subtracting 3 from the mv reading then dividing by 1200 to get a percentage. Better still would be something like a map/dict with voltage:percentage relationships.

## Things to do

If we go this route, we would need to cover the following:
- [ ] implement `getBattPercentage`
- [ ] add battery chemistry curves
- [ ] update the UI in firmware to pull percentage instead of guessing
- [ ] other apps and closed firmwares would need to update as well

oltaco, 2025-06-17:
Isn't mV battery-agnostic? Currently firmware just reports mV and it's up to clients to interpret that how they wish. Maybe battery type profiles would be better suited to live in clients?

446564, 2025-06-17:
The only issue is some clients are the device, so the device itself needs a way to know what charge is remaining.

446564, 2025-06-17:
I think adding a battery percentage function will not add a lot of overhead, and then it can be right there in the telemetry by default with the voltage.

closed 2026-10-11 by github-actions[bot]