meshcore-dev/MeshCore #460: Add radio support to firmware and configuration for app to include >= 920MHz settings for Japan

truedat101, 2025-06-29:
The request is based on the FAQ information about supported hardware https://github.com/ripplebiz/MeshCore/blob/main/docs/faq.md#121-hardware

specifically it mentions the supported frequencies topping out at 915 MHz.

Because the information for Japan is a bit obscure, I dug around and found a recent PR that fixes this set of problems on Meshtastic:

https://github.com/meshtastic/firmware/pull/4446

If the team needs a PR, we can do that.  If you can consume the above link to see how Meshtastic resolved this, that's great too.  My community here working on this we call "offline SNS" (conceptually meaning an offline social network ... easier to explain than mesh networks) will be able to provide PRs for mobile app work as well to handle localization (I assume you guys are using flutter or react native).  We can file a ticket for localization however, English will suffice as long as we can sort out settings for Japan.   The defaults aren't an option in the mobile app, so it follows I guess, firmware changes, and mobile app changes are needed after.

If a PR is wanted, please point me in the right direction on source, and I'll get this in this week and test.

recrof, 2025-07-06:
@liamcottle can add Japanese preset to the app, you just need to provide which frequency tx/power is legal in the region.

truedat101, 2025-07-06:
Let me dig through the docs on Japaan.  I may not have all of the terminology figured out, so I may need a hand in clarifying specs.

liamcottle, 2025-08-06:
> Sounds good. I think the main thing is ... I need to dig through the expected settings for meshcore, and map in we need for Japan. Is the app not open source? I'd prefer to submit a PR ... but somehow I seem to remember the apps are closed source.

Yeah, the mobile app is currently closed source. But the suggested radio settings are actually stored in a JSON config file on my API server, so nothing actually needs changing in the app to get the settings visible to users.

> I provided a link to the expected settings for Japan in the description. For brevity, here is the table below. Is it possible to just start the conversation in GH and not on discord. Because this isn't a Japanese originating project, you won't get a lot of people from Japan submitting requests because of the language barrier, though we now have two regions in Japan with people interested in this project.

No worries. Feel free to follow up here anytime.

> What other information is required ?

The mobile app just has a list of "Suggested" presets. Usually just one radio setting that's actually used in a given area. If you can provide us with the following information, it can be added as a suggested preset in the app:

- Frequency
- Spreading Factor
- Bandwidth
- Coding Rate (usually default of 5)

At this stage, we don't plan to have drop downs or selectors for the entire legal frequency range on a per region basis. We just have a list of options that the community are actually using right now.

So if you already have an active MeshCore mesh, and want to get others to join, the easiest way is to have those radio settings show up in the app as the suggested default.

If you don't already have these settings, you'd need to choose them yourself, or collaborate with others in your region to decide what settings you want to use.

Once you have them, let us know, and it can be added to the app.

truedat101, 2025-08-06:
Great, I will post to the group and see if there is consensus on the network settings.