meshcore-dev/MeshCore #461: RAK4631 + RAK12500 GPS not initializing on MeshCore v1.7.1

Hiltronix, 2025-06-29:
Hardware:
    Core: RAK4631
    Baseboard: RAK19003
    GPS: RAK12500 (Installed in Slot 3)
    Display: RAK1921 OLED

Firmware:
    MeshCore v1.7.1

Issue:

I installed MC v.1.7.1 today using the online FW flasher.
So far I haven't had any luck with GPS working on this hardware combination, so I was simply waiting till a version was released for the 4631 with GPS.
The change log for 1.7.1 says "GPS and sensor support added: Heltec V3, T-Beam supreme, T-Lora v2.1, RAK4631".
But for me, GPS is not working with the RAK4631 and the RAK12500, at least not when installed on the Baseboard RAK19003.
I believe there is only one slot on the RAK19003 that the RAK12500 GPS can be installed, and I believe it is Slot C, which is where mine is installed.
I know that GPS worked when I had Meshtastic installed on this unit, before switching to MeshCore.

RAK Wireless has several different baseboards, and I think some have more than one slot position where the RAK12500 could possibly be installed.
Can someone tell me which baseboard was used for testing, and was the GPS module the RAK12500, or was it some other GPS module that uses an IO header that isn't a WisConnector?

If getting GPS working on the various 4631 baseboards and/or WisBlock slots is a matter of tweaking the code for each one, I'm happy to test for you guys on this HW combination, if that helps at all.

Thanks for this great project!

Carey

prettyfly26, 2025-07-01:
I am having the same issue with the RAK 12500 GPS and RAK4631. Same specs as above.

WJ4IoT, 2025-07-02:
As far as I know, but please mind me I am a NOOB/Newbee in this respect, you should compile your own "pio run -e RAK_4631_GPS_companion_radio_ble"

However I do not know you have done this already, I do not have a GPS sensor myself so I cannot verify this will be your solution, I do know I have at the moment similar problems to activate the BME680.

Hiltronix, 2025-07-03:
Thanks @Defeator2020 for pointing out it's available in the repo to install manually, or install as a "Custom Firmware" using the Web Flasher, if you download it first.  Not intuitive or convenient, but now I know what I have to do to get it installed.  The GPS version also doesn't appear in the drop-down for the Download button on the Web Flasher page for the RAK4631.

I installed the GPS version from the repo list, and the GPS feature works fine on my RAK4631 with RAK12500.

I wonder why there are 2 versions, and not a check in the firmware if it has GPS, if not it disables the feature.  There must be a good reason not to do it that way.

Ideally, if there can't be a FW check for GPS, the Web Flasher page should have both versions listed for the RAK4631.  One entry for the GPS version, and one entry for the non-GPS version.  Otherwise I would imagine this confusion will happen to others.

Thanks again.