meshcore-dev/MeshCore #472: T1000e (+other LR110 radio?) ACKS not seen by sender on DM

seagull9000, 2025-07-02:
Possibly still same issue with T1000e under 1.7.2 for me.
Heltec Wireless Tracker (v1.7.1 or v1.7.2) and T1000e (v1.7.1 or v1.7.2) ...  both direct path
T1000e -> HWT: Message shows up in HWT and T1000e shows "delivered" immediately.
HWT -> T1000e: Message shows up in T1000e immediately, but HWT continues sending for 5 tries, then eventually shows "delivered".
It may be possible that in the latter case the path changed to flood for the final try, but the path on the HWT remained direct after the exchange.
This is reproducable every time.

Above is replicable with Thinknode M1 -> T1000e.

Setting path from HWT -> T1000e to "flood" works first time.  So issue is with DM, it seems.

I'll add some info from debug log later today.

fdlamotte, 2025-07-02:
another difficult one ...

;)

will try to reproduce, heltec v3 hw is similar to tracker no ?

seagull9000, 2025-07-02:
> 
> will try to reproduce, heltec v3 hw is similar to tracker no ?

Yes - works off same base

seagull9000, 2025-07-02:
Here is a log of the debug output on the T1000e:

```
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -96
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -96
: RX, len=22 (type=2, route=D, payload_len=20) SNR=7 RSSI=-45 score=1000 hash=48943953AAFA8040 [D1 -> AF]
BLE: writeBytes: sz=25, hdr=136
BLE: writeBytes: sz=1, hdr=131
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=22, hdr=16
: TX, len=6 (type=3, route=D, payload_len=4)
DEBUG:  Dispatcher::loop(): WARNING: outbound packed send timed out!
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
DEBUG: RadioLibWrapper: noise_floor = -96
DEBUG: RadioLibWrapper: noise_floor = -95
: RX, len=22 (type=2, route=D, payload_len=20) SNR=9 RSSI=-51 score=1000 hash=2AA14F09DCBA9F06 [D1 -> AF]
BLE: writeBytes: sz=25, hdr=136
BLE: writeBytes: sz=1, hdr=131
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=22, hdr=16
: TX, len=6 (type=3, route=D, payload_len=4)
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
DEBUG:  Dispatcher::loop(): WARNING: outbound packed send timed out!
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
: RX, len=22 (type=2, route=D, payload_len=20) SNR=8 RSSI=-47 score=1000 hash=9C1B1AF4862A23A7 [D1 -> AF]
BLE: writeBytes: sz=25, hdr=136
BLE: writeBytes: sz=1, hdr=131
DEBUG: RadioLibWrapper: noise_floor = -95
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=22, hdr=16
: TX, len=6 (type=3, route=D, payload_len=4)
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
DEBUG:  Dispatcher::loop(): WARNING: outbound packed send timed out!
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
: RX, len=22 (type=2, route=D, payload_len=20) SNR=9 RSSI=-45 score=1000 hash=E4E19EE01D648137 [D1 -> AF]
BLE: writeBytes: sz=25, hdr=136
BLE: writeBytes: sz=1, hdr=131
: TX, len=6 (type=3, route=D, payload_len=4)
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=22, hdr=16
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
DEBUG:  Dispatcher::loop(): WARNING: outbound packed send timed out!
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
DEBUG: RadioLibWrapper: noise_floor = -95
DEBUG: RadioLibWrapper: noise_floor = -95
: RX, len=22 (type=2, route=F, payload_len=20) SNR=9 RSSI=-52 score=1000 hash=CDEA707FEE10A858 [D1 -> AF]
DEBUG:  Dispatcher::checkRecv(), score delay below threshold (0)
BLE: writeBytes: sz=25, hdr=136
BLE: writeBytes: sz=1, hdr=131
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=22, hdr=16
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
BLE: readBytes: sz=1, hdr=10
BLE: writeBytes: sz=1, hdr=10
: RX, len=23 (type=2, route=F, payload_len=20) SNR=-1 RSSI=-55 score=1000 hash=CDEA707FEE10A858 [D1 -> AF]
DEBUG:  Dispatcher::checkRecv(), score delay below threshold (0)
BLE: writeBytes: sz=26, hdr=136
: TX, len=22 (type=8, route=F, payload_len=20) [AF -> D1]
DEBUG: RadioLibWrapper: noise_floor = -110
: RX, len=22 (type=8, route=D, payload_len=20) SNR=9 RSSI=-51 score=1000 hash=F49215600FAB2676 [D1 -> AF]
BLE: writeBytes: sz=25, hdr=136
BLE: writeBytes: sz=33, hdr=129
BLE: readBytes: sz=33, hdr=30
BLE: writeBytes: sz=148, hdr=3
BLE: readBytes: sz=33, hdr=30
BLE: writeBytes: sz=148, hdr=3
DEBUG: RadioLibWrapper: noise_floor = -97
DEBUG: RadioLibWrapper: noise_floor = -96
DEBUG: RadioLibWrapper: noise_floor = -96
DEBUG: RadioLibWrapper: noise_floor = -95
BLE: readBytes: sz=1, hdr=20
BLE: writeBytes: sz=11, hdr=12
DEBUG: RadioLibWrapper: noise_floor = -95
```

seagull9000, 2025-07-02:
@fdlamotte I've updated the fix in the original message.

In Dispatcher.cpp, the following change works (tested on T1000e):
```
uint32_t max_airtime = _radio->getEstAirtimeFor(len)3/2;
```
to
```
uint32_t max_airtime = _radio->getEstAirtimeFor(len)5/2;
```

I haven't investigated the root cause, so I don't know if this change is the permanent fix, but it will affect all devices.
If a PR is to be created, maybe ringfence it with a #ifdef LR110 ??

I'll leave this open for further investigation if needed.

fdlamotte, 2025-07-03:
Ok, could reproduce the bug

Could verify the 5/2 workaround

But also, when applying the LR1110-Workaround branch this works for me, really thought this had been included in 1.7.2

https://github.com/ripplebiz/MeshCore/tree/LR1110-workaround

@seagull9000 can you verify the workaround does its job for you ?

@ripplebiz do you think we should take care of all the protocols handled by the LR1110 and that we don't use ? It would be a switch/case over the new calculations you propose where we would put the original LR1110 code, or simpler (and dirtier) a call to the original code if protocol is not LORA ;)

seagull9000, 2025-07-03:
Thanks all