meshcore-dev/MeshCore #485: Full Duplex / Is MeshCore deaf while transmitting?

michaelhart, 2025-07-04:
I've noticed unreliability while testing MeshCore if two clients are trying to send messages at the same time. It made me wonder if it's capable of full duplex communication, or if while my client is transmitting (or repeater is repeating), if they are fully deaf and will miss data? My experience playing with a few clients and repeaters leads me to believe it's half duplex and totally deaf while transmitting, making mesh availability rather poor - especially for critical nodes which connect a lot of other clients/repeaters.

It seems like the stats for "total airtime" could also be summarized as "downtime"/or "time not listening" if so.

recrof, 2025-07-04:
you can't have full duplex on radio that uses only one frequency. all cheap LoRa radios(sx12xx, lr1xxx) are half duplex. collisions happen - we try to get around that by listening before tx, but it's never 100%.

446564, 2025-07-05:
Also without the frequencies being fairly far apart and using filtering to block either side, they are going to be deaf while the other is transmitting.

michaelhart, 2025-07-05:
Thanks for the replies! Please forgive my ignorance - radio and hardware generally aren't my areas of expertise. Part of me wonders how then WiFi works by comparison, and what would be needed to achieve the same level of apparent reliability for packet delivery as say a WiFi connection. Digging into this a little with ChatGPT, assuming it's correct, WiFi is also half-duplex and uses only a single frequency at a given time per connection. Yet that limitation is entirely removed from the user's concern, and we can upload and download simultaneously - the radio just switches frequently and overall manages everything without any data loss.

In the Seattle area, there's 1 node on Cougar Mountain that connects a lot of people in the area: https://pugetmesh.org/meshcore/equipment/

Without this one particular node, a lot of people in the PNW wouldn't be able to connect in the current state of the mesh. Based on my own testing with several clients/repeaters, my expectation is that the more the network grows, the more this one particular node is going to be deaf more and more % of the time.

So maybe full duplex isn't the right request, but is there anything that could be done to drive transmission reliability so that increased utilization doesn't come at the direct cost of availability/reliability?

> all cheap LoRa radios(sx12xx, lr1xxx) are half duplex

And in the same vein as above, are there more expensive radios which would be better suited for repeaters so that they can perform better?

recrof, 2025-07-05:
there are more expensive radios for concentrator nodes, but those get around the issue by using mutiple frequencies(which cheaper sensor radios can't utilize). also, concentrator radios are black boxes with no RadioLib support.

closed 2025-08-31 by liamcottle