meshcore-dev/MeshCore #492: Can’t log into my repeater from my T1000-E device.

gdurso53, 2025-07-05:
Can’t log into my repeater from my T1000-E device. I’m using the correct Password that I entered from my USB Web Client but every time I enter the password fails to connect. I’m the repeater and the T1000-E are in the same room??  So seems like it should work. 

Running the latest software  1.7.2 on  T1000-E and 1.7.0 on the T-Beam repeater.

recrof, 2025-07-05:
please try to flash this firmware via https://flasher.meshcore.dev/ - last option: Custom firmware. (don't unzip it, upload it as is)

[t1000e_companion_radio_ble_max_airtime_fix.zip](https://github.com/user-attachments/files/21081101/t1000e_companion_radio_ble_max_airtime_fix.zip)

gdurso53, 2025-07-07:
Exactly how to I instal flash this particular software on my device?

gdurso53, 2025-07-10:
There were 3 types of T-beam listed. I used the 1.2 (SX1262). It did not have a 1.7.2 version. The highest version was 1.7.0. There was a 1.7.2 for the one listed as (SX1262). When I installed the 1.7.2 the T-beam would not allow a usb connection. Then tried the 3rd option for Supreme (SC1262). But this version repeater 1.7.2 also would not let me connect via usb. So it seems to me the only option for me was the Lilygo 1.2 (SX1276) with repeater v1.7.0 

Do you know why this one didn't have an option for v1.7.2?

recrof, 2025-07-10:
ok, seems that the file names for tbeam sx1276 have changed, so they did not get into the flasher. it's fixed now, you can flash your tbeam to newest version.

oltaco, 2025-12-02:
This is complete, let's close it @liamcottle @recrof