meshcore-dev/MeshCore #497: Double Press generates Zero Hop Advert - should be Flood Advert

MikesAllotment, 2025-07-06:
The current double press button handler in Companion Radio generates a Zero Hop Advert - should be Flood Advert.

bool MyMesh::advert() {
  mesh::Packet* pkt;
  if (_prefs.advert_loc_policy == ADVERT_LOC_NONE) {
    pkt = createSelfAdvert(_prefs.node_name);
  } else {
    pkt = createSelfAdvert(_prefs.node_name, sensors.node_lat, sensors.node_lon);
  }
  if (pkt) {
    sendZeroHop(pkt);   // **<<----**  Wrong
    return true;
  } else {
    return false;
  }
}

Change to ....

bool MyMesh::advert() {
  mesh::Packet* pkt;
  if (_prefs.advert_loc_policy == ADVERT_LOC_NONE) {
    pkt = createSelfAdvert(_prefs.node_name);
  } else {
    pkt = createSelfAdvert(_prefs.node_name, sensors.node_lat, sensors.node_lon);
  }
  if (pkt) {
    sendFlood(pkt);   // **<<----**  Should be
    return true;
  } else {
    return false;
  }
}

recrof, 2025-07-06:
> I've never raised a PR - wouldn't know where to start.

it's pretty easy.. switch the branch to dev, find the file that needs to be changed, use the pencil to edit the file, make the changes and then submit it - PR creation will be one of the options.

seagull9000, 2025-07-06:
Why should it be flood? 

There must have been some logic for it to be zero advert.  Does this need some discussion?

MikesAllotment, 2025-07-06:
I was supposed to be flood ... it's a bug.

446564, 2025-07-12:
Can be closed? @ripplebiz