meshcore-dev/MeshCore #509: Use compression

mikecarper, 2025-07-10:
Some ideas:  

https://github.com/atomicobject/heatshrink

https://github.com/nclack/arithcode/compare/master...akohlsmith:arithcode:cleanup

https://github.com/gzm55/shoco

https://github.com/rjohnsondev/smaz

Could also look into a pre shared library for the protocol to compress the standard headers being sent.

mikecarper, 2025-07-10:
Based off the projects linked, compression is useful for just about for messages that contain any number of words. Compression of the message would be done before encryption. Do a strlen comparison to make sure compression of the message was the correct choice before sending; if the same do not use compression.

I'd have to look at the header bits but there might be a pattern for different regions that could benefit from compression as well. This would be a hand picked library for the most common configurations though. 

Would require 2 bits; flag for header compression and flag for message compression.

recrof, 2025-07-11:
we tried to toy with the idea of compression and in 160 characters there is little to no gain. if you got different results, please share.

mikecarper, 2025-07-11:
I'm going to see if I can get some test data from the local MQTT server and report back how each project does with that data

jbrazio, 2025-07-12:
This is what you need to show to sell the idea.. :-P

Less data, less time on air, this could potentially reduce between 30-50% air-time.
Ok it will not be linear because we have more data on each packet.

[image]

We would need some real world data to be able to measure the implementation complexity vs the gain but in a nutshell if we can archive 30% less air-time on average.. stfu and take my money.

@ripplebiz comments ?

mikecarper, 2025-07-12:
One concern is hardware power. On devices that have a screen, those usually have more capacity. On the simple ones, compression and decompression can be done in the app

jbrazio, 2025-07-12:
Never on the app, this needs to be firmware. Even without screen they use esp32, rp2040 or the nrf.. all of them should be capable of dealing with this.

mikecarper, 2025-08-26:
Been thinking about this and for the English language you could do 6bit encoding which would get us just over 240 char per message or 5bit (lowercase and some punctuation) for just shy of 300 char per message 

2 bits to say what kind of encoding is being used.
Full utf8
5bit
6bit
Compressed

jbrazio, 2025-09-16:
IMO this is not the right thinking, we should not aim to send "more text", the aim of compression is to send the same amount of text.. faster thus reducing air time and network congestion.

@liamcottle @mikecarper