meshcore-dev/MeshCore #519: Heltec Wireless Paper hangs at on reboot (post flash) with "Please wait..."

ActuandoWoody, 2025-07-12:
Heltec V3 Paper hangs on boot - post flash with Heltec_Wireless_Paper_companion_radio_ble-v1.7.2-0963341-merged.bin and Heltec_Wireless_Paper_companion_radio_ble-v1.7.2-0963341-merged.bin

Flashes fine (and seems to operate fine) with Heltec_WSL3_companion_radio_ble-v1.7.2-0963341.bin and any/all Meshtastic fw (for the HV3 Paper).

That's it really. I know this is the first Meshcore release of fw dedicated for this device, but it doesn't work.

ActuandoWoody, 2025-07-13:
> can you provide revision version you have? could help the author of the port with debugging.

Happy to - if you can tell me how to find that?

Nothing obvious on the board, and blowing the firmware reports:

VendorID 0x10c4 
ProductID 0xea60

recrof, 2025-07-13:
You're right - the device does not work from flasher.. please unzip and use web flasher's "Custom Firmware" option.
merged one is for clean install.
[Heltec_Wireless_Paper_companion_radio_ble-1.7.2-dev.zip](https://github.com/user-attachments/files/21204526/Heltec_Wireless_Paper_companion_radio_ble-1.7.2-dev.zip)
I'll open a pull request to fix this in the next release

recrof, 2025-07-13:
https://github.com/ripplebiz/MeshCore/pull/522

ActuandoWoody, 2025-07-13:
> You're right - the device does not work from flasher.. please unzip and use web flasher's "Custom Firmware" option. merged one is for clean install. [...]

Brilliant - thanks!

I used Heltec_Wireless_Paper_companion_radio_ble-1.7.2-dev-merged.bin from the zip you posted - and that has worked a treat.

liamcottle, 2025-12-02:
> I guess this issue is fixed now and we can close it?

Sounds good!