meshcore-dev/MeshCore #549: Preset Names and Coding Rate settings

mikecarper, 2025-07-23:
Narrow Bandwidth seems to be the winner based off the testing we're doing in the Seattle Mesh. I propose we have some presets that can be used to make setting changes easier to do. 

| Preset Name | Spreading Factor | Bandwidth kHz | kbps | dBm    | Message Air Time (s) |
|-------------|------------------|---------------|------|--------|----------------------|
| ${\textsf{\color{brown}Brown}}$        |               11 |          62.5 |  0.3 | -137.5 |                 3.55 |
| ${\textsf{\color{red}Red}}$         |               10 |          62.5 |  0.5 | -135.0 |                 1.94 |
| ${\textsf{\color{orange}Orange}}$      |                9 |          62.5 |  0.9 | -132.5 |                 1.05 |
| ${\textsf{\color{yellow}Yellow}}$      |                8 |          62.5 |  1.6 | -130.0 |                 0.59 |
| ${\textsf{\color{green}Green}}$       |                7 |          62.5 |  2.7 | -127.5 |                 0.32 |
| ${\textsf{\color{blue}Blue}}$        |                7 |           125 |  5.5 | -124.5 |                 0.16 |
| ${\textsf{\color{purple}Purple}}$      |                7 |           250 | 10.9 | -121.5 |                 0.08 |
| ${\textsf{\color{violet}UV}}$          |                7 |           500 | 21.9 | -118.5 |                 0.04 |

To get the "Message Air Time in seconds" column I used this calculator https://www.semtech.com/design-support/lora-calculator and this test message which is 87 bytes  
`"The quick brown fox jumps over the lazy dog.", "The five boxing wizards jump quickly."`
 
Equitant speed to 11_250 or what's known as "Long Range / Fast" is the "Orange" preset. I propose that the "Yellow" or "Green" preset would be the new default. We're using the "Green" preset currently here in the Seattle Mesh. 

Adding on frequencies to the color is how to communicate the full preset OR we pick some default frequencies to go with each color.   
Green 910.525 OR Green (which has the setting of 910.525)

<hr>

And then with the Coding Rate we should set everyday carry units like the t1000e to 4/8 and repeaters to a default of 4/5. Make this change easier to find in the app so it can be adjusted if needed. The coding Rate can be changed and the radios can still talk to each other on the same preset. In app say that you can adjust the Coding Rate to get more or less range for hop 0.

446564, 2025-07-23:
Heck yeah

mikecarper, 2025-07-24:
If we ever want to cutoff older hardware then the presets might change again as more narrow bandwidths are going to be more desirable in theory https://github.com/ripplebiz/MeshCore/issues/544

inversenebula, 2025-07-24:
Unnecessary additional confusion which is not self-explanatory, please don't do this.

mikecarper, 2025-07-24:
62.5 is on the edge of what can be done without a tcxo if the messages are short enough.

Uroboros67, 2025-08-25:
Hello
If a narrow band 62.5kHz has been introduced, instead of sticking to the middle frequency of 869.525, it is not better to introduce something like 4 channels with the middle frequencies:
Ch.0 - 869,43125
Ch.1 - 869.49375
Ch.2 - 869.55625
Ch.3 - 869.61875

Regards Jacek