meshcore-dev/MeshCore #551: Support more than 100 contacts

michaelhart, 2025-07-25:
With the 100 contact limit in the PNW, I'm now at a point where I'm having to drop reachable repeaters. I know some hardware that's commonly available has MBs of storage, so the 100 contact limit seems like it could scale much higher on hardware where the storage is greater.

It'd be a huge win for users in areas with healthy and quickly growing meshes who I think would rather not have to juggle managing which repeaters to add or remove plus benefit from maintaining all of the known repeaters on the device for path/signal testing and path analysis.

446564, 2025-07-25:
Totally agree, we host need to figure out a way to expose that extra storage.

recrof, 2025-07-25:
working on porting xiao mg24 which is as energy efficient as nrf52 and has ton of extra storage.

mrlmuk, 2025-07-27:
I have been increasing the limit to 140 in the code and it seems to work fine. Issues which were present before this increase or the number of contact past 100 have not been made any worse. It could perhaps best be managed with a combination of the app managing some of the contact functions in a separate address book and an increased limit in the code.

The storage at least on the T1000-E only shows 28kB though and I am at 89% of that with 120 contacts. I'm not sure where that number of 28kB comes from though as the device has 1MB I believe and the firmware doesn't appear to be that big.

Also mentioned https://github.com/ripplebiz/MeshCore/issues/390 https://github.com/ripplebiz/MeshCore/issues/39

446564, 2025-08-06:
This is underway by @oltaco and so far testing well.

qaylinmoore, 2025-08-14:
@mrlmuk did you ever figure out where that 28kB number comes from? I've been trying to hack around with that on my RAK4631? It feels like I should be able to override it in LittleFS. I cut a bunch out of my meshcore build just to see if the space available went up, but it didn't.

446564, 2025-08-14:
> [@mrlmuk](https://github.com/mrlmuk) did you ever figure out where that 28kB number comes from? I've been trying to hack around with that on my RAK4631? It feels like I should be able to override it in InternalFS. I cut a bunch out of my meshcore build just to see if the space available went up, but it didn't.

It's a predefined user data partition. Part of the board soft device or something like that. Not something easy to adjust.

haylinmoore, 2025-08-14:
> It's a predefined user data partition. Part of the board soft device or something like that. Not something easy to adjust.
I was just talking to oltaco about it, and that matches. They're working on a way to use the application data section https://github.com/oltaco/MeshCore/tree/CustomLFS, so that should be cool

446564, 2025-08-14:
Haha ya my t1000e and nano g2 are both running this.

mrlmuk, 2025-08-28:
I knew it would be something from the base framework but didn't find a simple solution in the code. I decided that this device doesn't need 100 contacts as you will see errors if you export config over Bluetooth well before that.  Purging contacts and setting it to 60 solved the storage and corrupt config exports. 

The same problems occur with the Xiao nrf52

mrlmuk, 2025-08-28:
Yeah I saw that and it's probably going to be needed to maintain feature parity between ESP and NRF devices in future. This small partition is common to Seeed Xiao nRF and maybe others.

I have also seen somewhere in the branches that the contacts limit is being increased as I have been doing on my other devices, it's a simple change to the variant platformio.ini file.

liamcottle, 2025-09-16:
Most variants now have increased contact capacity. Closing as completed.

closed 2025-09-16 by liamcottle