meshcore-dev/MeshCore #567: [Feature Request] Display rotation for different cases.

mrlmuk, 2025-08-01:
Particularly on the Heltec LoRa 32 v3, add the ability in code to set the scree rotation, at least between 0/180. This will allow cases to be designed with the buttons and USB port on the top.

Devices such as phones pretty much always put the USB port on the bottom.  Other than the antenna traditionally being at the top there is no good reason for this tradition. It means you cannot hold in some positions, or stand it up with the cable in and the port gets debris in it more often. Thankfully some tablets are now changing the position to the upper side away from hands.

While some will think it's bad idea they don't have to use it. Meshtastic and rnode both already have this feature so it might be an easy one to implement with minimal code.

recrof, 2025-08-01:
look at DISPLAY_ROTATION compile flag

recrof, 2025-08-01:
if you look at the ST7735Display.cpp you will be able to implement the same behaviour in SSD1306Display.cpp with trivial changes.

mrlmuk, 2025-08-01:
Ok I got this working with a little help from Copilot. I knew what needed adding not exactly how to change the begin section.

In \src\helpers\ui\SSD1306Display.cpp I added this block...

#ifndef DISPLAY_ROTATION
  #define DISPLAY_ROTATION 0  // Options: 0, 1, 2, 3
#endif

& replaced the existing block for begin...

bool SSD1306Display::begin() {
  bool success = display.begin(SSD1306_SWITCHCAPVCC, DISPLAY_ADDRESS, true, false) && i2c_probe(Wire, DISPLAY_ADDRESS);
  if (success) {
    display.setRotation(DISPLAY_ROTATION);
  }
  return success;
}

In \variants\heltec_v3\platformio.ini I added the flag
[env:Heltec_v3_companion_radio_wifi]
 -D DISPLAY_ROTATION=2

When it didn't work at all I went back and added it to the correct variant, and a third time I decided to count from 0 instead of 1. Now the display is upside down as I wanted. If this all seems sensible I can try to figure out how you submit the code on here but I have never done that, I just copy locally and edit directly normally.

liamcottle, 2025-08-31:
It looks like you got this working for you own firmware build. If you'd like to see it merged into official firmware, please open up a PR with the proposed changes. I'll close this one out since you got it working in your build.

closed 2025-08-31 by liamcottle