meshcore-dev/MeshCore #568: [New HW] Add support for meshimi

alexbegoon, 2025-08-02:
Hi,

I’d like to add support for [Meshimi](https://github.com/alexbegoon/meshimi) to this project. I'm happy to implement it myself, but I wanted to align with the maintainers first so it doesn’t come as a surprise with a big PR 🙂

Would that be okay? Any concerns or suggestions before I start?

Thanks!

recrof, 2025-08-02:
it should be pretty straight-forward if you use xiao c6 as a starting point. would say that you only need to edit the pins in platformio.ini and you're done. I guess you have some extras on the board, so that'll be the bulk of it. if not, I would not even create special variant, just add new env into xiao c6 platformio.ini.

alexbegoon, 2025-08-02:
I see, thanks @recrof !

closed 2025-08-09 by alexbegoon