meshcore-dev/MeshCore #574: nrf52840 (ProMicro) + LLCC68 does not work

ihugof, 2025-08-03:
Hi There, 

Compiled the code for ProMicroLLCC68_Repeater, it installs OK, but after that nothing - no console, no way to configure the repeater, nothing on serial port as well.

Also tried the Companion_Radio_BLE and USB and noting - no bluetooth or usb.

Something is getting stuck somewhere...:)

BTW...tried MeshTastic firmware just to check if it could be something with the ProMicro or LLCC68, but it works fine there. 

Please check!

Thank you!

HF

recrof, 2025-08-03:
hello, this variant is getting removed because we don't have developer that is able to actively maintain it and there is very little interest. it was broken for 2 months and nobody noticed. if you are interested in fixing it and maintaining it, we are ready to re-introduce it back to the flasher.

oltaco, 2025-08-04:
This is going to sound extremely stupid but have you tried using the standard faketec/promicro firmware?

The reason I ask is that apparently some LLCC68 chips actually report as sx1262 anyway and one of our users has had luck using the normal firmware because of it.

I actually have an LLCC68 chip here on my bench but I haven't had a chance to solder it up so I can get these builds working again.

ihugof, 2025-08-05:
@oltaco No worries but yes, I did try it first, no luck, then saw on Git that there was a specific version for LLCC68, compiled it, installed it and nothing - the result it's the same.

Actually you can try just the ProMicro itself connected to the USB, without any radio chip and the behavior it's similar - no configuration possible and from the serial I just get a "failed init", but bluetooth for client or USB for repeater should work and allow it to be configured anyway, but no.

With the radio chip, in this case the LLCC68, no "failed init" message but bluetooth dead when client and usb dead on repeater, so it's basically the same.

Any way to troubleshoot this?

ihugof, 2025-08-05:
@recrof You mean you are going to remove the ProMicro nrf52 (FakeTec) all together or just the LLCC68 variant one?

Unfortunately I'm no developer :(

recrof, 2025-08-06:
faketec with sx1262 is very supported by @oltaco, it won't be removed. I'm only talking about llcc68 variant.

ihugof, 2025-08-06:
@recrof That's nice to know! :) Maybe @oltaco can find some time to also dedicate to LLCC68! ;) One can only hope! :)

LLCC68 chips cost around 50% less than the SX ones, at the cost of some "earing loss", but that would help to expand the network.

lincomatic, 2025-08-19:
The LLCC68 build has been removed, because it actually works with the current SX1262 build

ihugof, 2025-08-20:
Hi @lincomatic , thx for the reply!

I installed the firmware but it does not do anything - bluetooth doesn't come on, no message on terminal, nothing.
To make sure this was working, I tried flashing Meshtastic and it works fine, radio chip is detected and bluetooth works, but when I flash Meshcore, nothing

What's the best way to debug it? If you point me in the right direction, maybe I can help. :)

Thank you!

lincomatic, 2025-08-21:
> Hi [@lincomatic](https://github.com/lincomatic) , thx for the reply!
> 
> I installed the firmware but it does not do anything - bluetooth doesn't come on, no message on terminal, nothing. To make sure this was working, I tried flashing Meshtastic and it works fine, radio chip is detected and bluetooth works, but when I flash Meshcore, nothing
> 
> What's the best way to debug it? If you point me in the right direction, maybe I can help. :)
> 
> BTW I tried both companion USB and Bluetooth, and Repeater, no luck.
> 
> Thank you!

Just to be sure... you're saying the Faketec_xxxx builds don't work on your Faketec w/ LLCC68?

lincomatic, 2025-08-24:
@ihugof 
OK, that's the identical LLCC68 and ProMicro I'm using, except mine is for 915MHz. And same pinout
P0.02-MISO
P0.09-RESET
P0.10-DIO1
P1.11-SCK
P1.13-NSS
P1.15-MOSI
P0.29-BUSY
VCC  -VCC
GND  -GND
Mine's also just connected directly w/ soldered jumpers.
Did you run the full erase file before flashing in MC?