meshcore-dev/MeshCore #576: New Companion Radio screen layout needed?

seagull9000, 2025-08-04:
I'm opening this issue for discussion as it may be time for a coordinated screen layout refresh for the companion radio.

There are a some of drivers for a redesign:

1.  GPS ON/OFF status, number of sats, HOP?
2.  Buzzer ON/OFF
3.  Portrait/Landscape orientations.

I think we may have three main screens (after startup):
a. the idle screen, that has the current time, logo,  Buzzer status, and extended GPS status
b. the message(s) screen
c. button press help screen

I think if we define the row numbers dividing screen information in a common config, then doing future changes will be easier.

Open for suggestions/discussion ...

LitBomb, 2025-08-04:
I agree with these.  I would like to add a couple more things.  

If there is room for the idle screen, I would like to see the firmware version number.  with many radios, I don't always remember which one is updated which one has not been updated.  To check the version, I have to turn the radio off and then back on to see the version number that is displayed for a couple seconds during boot.  Not a huge deal, but a good QOL for me and hopefully other users.

Second is when double pressing the user button for an advert, change the message from "advert sent" to "zero hop advert sent" to confirm it is not a flood advert.

seagull9000, 2025-08-05:
I'll start with the abstraction layer first, get those commits in with some simple stuff, then we can work on the layouts, button-screen response mappings in the future.

seagull9000, 2025-08-06:
I've added background document for comment.

https://docs.google.com/document/d/1dGo09v9F4nX9vnSraGx4sVQRUsFkVJEv2LEkpUp18JA/edit?usp=sharing

seagull9000, 2025-08-07:
Ah - sorry - I missed a step in the sharing.  Open for comments