meshcore-dev/MeshCore #578: Custom paring pin show on dispaly

janekx, 2025-08-05:
For secure you can setup individual custom pin, but this pin is shown on display, so not much secure. Custom pin should not show on display.

ViezeVingertjes, 2025-08-20:
What if this is addressed by having the PIN appear on the display for a limited time—say, 10 seconds—after startup and then disappear?

This would solve the PIN permanently showing and from a user experience perspective, if someone forgets or misses the PIN, a simple restart is far more convenient than connecting to a laptop to retrieve it via a serial connection. This approach also simplifies the code, as it wouldn't need to manage different states like 'fixed' or 'random'. (It now just knows pin, not if its custom or fixed, if i am not mistaken)

janekx, 2025-08-22:
> > In that device can be personal messges, if someone want keep it secure will setup custom pin. In case of lost it should need to be deleted and reflashed and not accessible by showing pin every restart.
> 
> Ah! yes, for such scenario i get your point. Not showing it would be the better solution in that case, but then i think it should log it to serial though, there must be some way to recover it imo.
> 
> (MeshCore currently does not show pin to serial on startup)

I think it should not be any way to retrieve, in case of choosing such privacy and safety is your issue you lost pin. Many devices not allow pin retrieve.

ViezeVingertjes, 2025-08-22:
> I think it should not be any way to retrieve, in case of choosing such privacy and safety is your issue you lost pin. Many devices not allow pin retrieve.

If your device get's in hands of someone else; they will be able to connect even without pin so not sure what the use-case is... Not showing it on the screen by default, i can understand, as then it would be somewhat too easy(?).... but other than that, its all a false sense of security.  I don't think an unrecoverable PIN is the right solution for this...