meshcore-dev/MeshCore #580: Voltage and Current sensors values not signed

wel97459, 2025-08-05:
This links back to [CayenneLPP](https://github.com/ElectronicCats/CayenneLPP) and signed values are not setup in the function [getTypeSigned](https://github.com/wel97459/CayenneLPP/blob/8b258532e0003f4325e985a2169728cbe03543ef/src/CayenneLPP.cpp#L556C6-L556C47) this is a bit of a problem for me watching charging current vs discharge current using a INA219.

wel97459, 2025-08-08:
Here the issue made on the CayenneLPP repo https://github.com/ElectronicCats/CayenneLPP/issues/60

wel97459, 2025-09-19:
PlatformIO has been updated to the new version  of 1.6.1 and has the change in it.

The platformio.ini will need to have the version changed.

liamcottle, 2025-10-23:
Thanks! I'll close this out :)

closed 2025-10-23 by liamcottle