meshcore-dev/MeshCore #600: Sign public messages with 1 byte

mikecarper, 2025-08-10:
Hash the message and send the first byte. Would need clients to auto advert once a day. Messages that don't match up with the hash will be flagged; messages from nodes without a public key will also be flagged but differently. 

Could have a request advert functionality built in if we want as adverts become more important now.

mikecarper, 2025-08-10:
1 byte prevents same name confusion from happening in the app; doesn't prevent hacked firmware from sending out 1 message that will pass. Each message gets hashed so the more messages that get sent the higher chance that the impersonator doesn't pass the 1 byte check.

mikecarper, 2025-08-11:
https://github.com/georgemakrakis/TESLA
I think reference needs 12 bytes and decent clocks. Can always cut the bytes down 

Paper on it
https://www.researchgate.net/profile/Peng-Ning-3/publication/234795195_Multilevel_mTESLA_Broadcast_Authentication_for_Distributed_Sensor_Networks/links/00b7d52b6c27265f59000000/Multilevel-mTESLA-Broadcast-Authentication-for-Distributed-Sensor-Networks.pdf

ripplebiz, 2025-08-11:
Very interesting. I shall have a read.
Would be great to put some level of authentication into the channels, even if just moderately secure.

ripplebiz, 2025-08-11:
I'm quite fascinated with this TESLA algorithm. I'm still a bit fuzzy on some of the details, but I think I can grasp the basic mechanism of using the key chains going backwards, and the keys being used forwards (in time). Am thinking it could be possible to do an on-demand authentication, ie. having a way for a node to request the last key from a sender, rather than sender periodically sending the key for each time window. Or even asking 'does anyone have the last key?' because anyone can verify it's a valid key by checking with the 'commitment' key (which could be put into sender's advert, and signed)

mikecarper, 2025-08-11:
http://juliastrings.github.io/utf8proc/ Might be the easiest way forward to alert about a potential impersonation in public chat

closed 2026-09-12 by github-actions[bot]