meshcore-dev/MeshCore #602: [Feature Request] Control External Notifications

ne0phaze, 2025-08-11:
Currently there is no way I’m seeing to control the external notifications for devices with buzzers or LEDs. 

Some devices have button press combinations or on/off switches but the phone app only allows control notifications sent to the phone.

Preposed request

Enable a software toggle for buzzer and LED lights to be either on or off

mikecarper, 2025-08-12:
Per user or group chat or room server settings for the notifications would be awesome. So play all the bells and whistles for a DM from someone important but do nothing for public chat

Marvin-YYC, 2025-08-22:
I would really like to see a feature like this. 

I don't see any way to add a message notification LED to my Heltec V3 with the current firmware or companion app.

My Xiao ESP32s3 nodes use the heartbeat LED for message notification on MeshCore.  The Heltec doesn't use the heartbeat LED on MC?! The white LED only flashes during a tx.  Nothing for message notification.

I don't see anything in settings for external notification devices like Meshtastic has setup.

Would be great to see it in the MeshCore app or added to the firmware.

Burny08, 2025-11-30:
Yes, I would also like to receive external notifications, please!

bert010, 2026-02-07:
yes please!

zomgugoff, 2026-09-05:
Really need this. The GAT562 30S's activity light goes into a loop when the internal capacity hits 89%. It's extremely annoying.

Blake-Latchford, 2026-09-18:
I think there are a few issues called out in this one.
1. People want to be able to toggle this in the app and/or don't know about triple click.
2. Triple click doesn't disable LEDs.
3. The triple click timing might be too tight.

The second problem seems pretty easy to fix assuming the project just wants triple click to do both. Maybe the user feedback above says that is what users expect? Its pretty hard to be certain, because some of the feedback is clearly talking about something other than the initial description by @ne0phaze.

Amoulier, 2026-10-28:
I have device-side notification controls working on Muzi Works SuperBase and would like feedback on the upstream shape before submitting feature PRs. This is a concrete implementation of part of the original request, rather than phone notification settings.

Proposed separate PRs, initially scoped to SuperBase:
- A persistent LEDs On/Off page. Off suppresses the green heartbeat and message flashes during the application. The charger-driven red LED and bootloader indicators are explicitly outside its control.
- Persistent buzzer policies: All, Off, Messages, System and DMs Only. Messages includes direct/channel messages; System excludes them; DMs Only leaves channel messages silent. Device alerts can be used while the phone is connected.
- Matching active/inactive page symbols using the existing Bluetooth page layout. The prototype's derived Google Material artwork retains its Apache 2.0 attribution.

The implementation is available in the [SuperBase fork](https://github.com/Amoulier/MeshCore/tree/59b6896a07a571ef6ce445290417ac5945d9fbf3). Physical checks covered On/Off persistence across resets, LEDs Off across an update, all three selective buzzer policies with real direct/private-channel messages and startup events, and the page icons. The temporary channel was removed after testing; no node keys or configuration exports are needed for the proposal.

Would maintainers prefer board-local preferences/pages first, or a shared notification policy? The existing triple-click behavior and other devices' defaults can remain unchanged. I would keep one feature per PR once the preferred direction is agreed.

The separate small fix for an active RTTTL melody continuing after mute is submitted as #3710.

mikecarper, 2026-10-28:
https://mikecarper.github.io/MeshCore/notifications/ I made this tool for my firmware; totally forgot about this issue

liamcottle, 2026-10-30:
Regarding the buzzer, https://github.com/meshcore-dev/MeshCore/pull/3603 has been merged.

This allows toggling the companion node buzzer preference via the companion CLI.

- `get buzzer`
- `set buzzer on`
- `set buzzer off`

v1.51.0+ of the MeshCore App will also have a new notification setting to toggle this in the UI, rather than having to triple press the user button on the node.