meshcore-dev/MeshCore #611: Default to correct frequency

zmitton, 2025-08-13:
https://github.com/meshcore-dev/MeshCore/blob/b0946b3f6b6c74188381a740524b12b7079c5824/docs/faq.md?plain=1#L177

If the recommended frequency is 910.525 then can we actually set that in the code? I would guess that a huge ratio of users end up on 915 thus hurting users that even get it right.

In this application where "network effect" is clearly one of the most important things, this is especially detrimental.

liamcottle, 2025-08-13:
The mobile app has 910.525 as the frequency for the USA radio presets.

The firmware defaults are set to the UK frequency.

If you need something different as a default set in the firmware for first use, you'd need to build your own firmware files with the frequency build flag set to what you want.

recrof, 2025-08-14:
869.525 is default in the code for all roles. it won't be changed to US preset. would make much more sense to have it set to 0 and force the user to choose it when first connecting to the node.

IoTThinks, 2025-08-19:
I'm thinking if the firmware can automatically detect location via phone and set frequency based on the GPS of the phone.

446564, 2025-08-22:
> I'm thinking if the firmware can automatically detect location via phone and set frequency based on the GPS of the phone.

The app does not have location access and likely won't in the near future.

closed 2025-08-31 by liamcottle