meshcore-dev/MeshCore #619: Support for native Linux devices (i.e. Pi Zero 2)

jpoisso, 2025-08-15:
Hello!

Is there a plan or roadmap to support native Linux devices in the future?

I have a Pi Zero 2W with Lora HAT that I was hoping to use as a Room Server node but it doesn't seem to have support yet.

For reference, Meshtastic provides some support here: https://meshtastic.org/docs/hardware/devices/linux-native-hardware/

trisweb, 2025-08-27:
Seconded, in my case using on the ClockworkPi uConsole with a LoRa expansion module would be very nice.

mikehill-notabot, 2025-08-27:
> Seconded, in my case using on the ClockworkPi uConsole with a LoRa expansion module would be very nice.

I was digging around the code looking for the same, uConsole

ices-dev, 2025-09-01:
i would love to use my Zero 2 with POE as Room Server :)

gretel, 2025-10-26:
me too would like to use zero 2 with POE. regards

gretel, 2025-10-31:
found some more recent developments:

- [Implementation of Meshcore in Python](https://github.com/brianwiddas/meshcore-pi)
- [A lightweight Repeater Daemon implemented in Python, built using the pymc_core library](https://github.com/rightup/pyMC_Repeater)

trisweb, 2025-10-31:
Yeah this is very close if not complete at this point.

Has anyone tried the Rpi client?

edkraus, 2026-03-14:
> I created an updated PR #2025 to continue working on #1137.
>
> I'll maintain a fork until this is merged upstream and try to regulary rebase the work so we don't lack behind. Fork: https://github.com/l5yth/meshcore-linux/tree/linux branch linux. See variants/linux/README.md for details on setting this up.
>
> POC Repeater: https://analyzer.letsmesh.net/nodes/repeaters?public_key=63E822CAEE5548B7CB891F002F032247C1F54B08CEC4E39EAD2450F5C06E6AE2

Thank you ... I think.  I don't know how to follow all the details in the development pages of Github.  I'm assuming this is in reference to my requests elsewhere for there to be working USB connection support in the Linux and Windows apps.  FWIW, I don't use any Pi devices, everything I use is a Linux distro (often Mint) on a pc and sometimes a vm.
Ed Krausf.t.n.m.p.

l5yth, 2026-03-15:
@edkraus I'm not referencing your issue. I am working on the main task described above. Please ask on the forum or reddit for help with your Linux/USB questions.

trisweb, 2026-03-15:
@edkraus you are getting notifications from a thread with dozens of people on it. Not every notification you receive is directed to you. We are trying to keep this thread about the topic it was opened for, no need to respond.