meshcore-dev/MeshCore #646: Support for Ethernet on RAK-based boards (e.g., RAK4631)

drewzh, 2025-08-20:
Currently, Meshcore does not support nodes that use Ethernet on RAK-based boards such as the RAK4631. Adding support would help close this gap and make the platform more suitable for fixed repeater stations.

For example, I’ve been running a node in my attic for over a year on Meshtastic, where it’s both powered and fed via Ethernet from my network switch. This setup enables:

- A streamlined repeater configuration for remote placement.
- Wired power backup directly from the switch.
- A more reliable networking backbone, which is especially valuable for fixed, powered stations.

Supporting Ethernet in Meshcore would make these kinds of deployments much more practical and robust.

Hiltronix, 2025-08-20:
I'd like to see this as well.

I did something similar.  I started with experimenting with Meshtastic, and then later switched to MeshCore.  I installed the RAK hardware in a weatherproof box on a mast that I put up on my roof below the antenna.  The hardware is connected by Ethernet to my network, and powered by a PoE injector.

The hardware I used:
RAK4631-9-WB-I (Core)
RAK19007-0-BB-N (Baseboard)
RAK13800-E-WB-N (Ethernet Block)
RAK19018-x-WB-N (Ethernet Block PoE Adapter)

The node on the mast now runs MeshCore, it's still powered by the Ethernet cable, but that is all the Ethernet connection is good for at this point, powering the device.  I've been hoping at some point the MeshCore project would evolve to allow communication over Ethernet as well.  But when I saw this post, I thought it was a good time to chime in.

I've been kicking around the idea of running USB up the mast to the box, but with a run that long I'd need to buy expensive USB with power injection hardware for either end, which greatly increases the size of the box on the mast, and the heat that is generated in the box.  And it turns out that USB with power injection for long runs typically requires the destination USB hardware to have the power injector, which is fine for office situations, but far from ideal on a mast, which would then require me to run a power cable as well.  My point is that USB to a box on a mast is far from ideal, the hardware is larger, generates more heat in a sealed box, and requires in most cases another cable for power.  And this is a problem that has already been solved with Ethernet/PoE tech, that is cheap, low heat, and small.

It would be great to see Ethernet capabilities in MeshCore for mast and roof mount nodes.  Ideally for updating FW at least, so I don't have to get the ladder out to upgrade my node. :)

liamcottle, 2025-08-31:
Hey, thanks for the suggestion. This sounds interesting.

At this stage, I don't think this is something we are looking to focus on right now. As far as I know, most of the popular LoRa boards don't have ethernet at all, and for the RAK boards you have to add an additional module.

I'll close this one out, but, if you manage to get this working, without requiring massive changes to the code case, we could explore this further and might be able to add it in.

If you're already running PoE to a weather proof box on your mast, I would probably suggest running something like a Raspberry Pi, with a RAK connected to it over USB running the USB companion firmware. This would allow for updating the firmware on the RAK board via the Raspberry Pi, and is already something we support as-is.

Even if you were to add ethernet support to the MeshCore firmware for the RAK module, I don't think you'd be able to do firmware updates over it.

closed 2025-08-31 by liamcottle

CoolAEW, 2026-01-21:
I also have a RAK node with ethernet I'd like to be able to manage over the network. I'm also thinking this could be used to set up relay sites with 2 or more radios using directional antennas that communicate over ethernet. Using pole mounted switches you could then easily power and manage multiple nodes that have great long distance coverage in multiple directions