meshcore-dev/MeshCore #666: [Proposed contribution] Implement power savings features on T114

taedryn, 2025-08-25:
I'm interested in contributing code which will save power, starting on the T114, but eventually applicable to many or all nRF-powered boards.  There are numerous features of the nRF52840 and SX1262 for reducing power consumption, which are not currently being used.  The invocation of these features on the nRF should have no impact on the functionality of the mesh: powering down unused peripherals, switching to a more efficient power source, etc.

The one option I've identified on the SX1262 might impact mesh functionality, and would require careful testing, which is switching to a lower-power mode most of the time, with occasional events to listen for incoming data.

There is a huge opportunity available in modifying the main loop to enable CPU sleep when there isn't pending work.  This is also much riskier, since it does impact the main loop, and pending-work signals aren't currently available in the core code.  For a low-traffic mesh environment, this would result in a massive power reduction (on the order of 1/5 to 1/10th the power consumption, maybe better), and even in a high-traffic environment like the Seattle area mesh, due to the low-speed nature of LoRa communications, the CPU necessarily spends a lot of time doing nothing at full speed, right now.  I suspect this could allow us to use 1/3 the power even in a high-traffic mesh.  An alternative/additional approach would be to enable lower-power clock modes, such as reducing clock frequency, without going into a full Wait For Event sleep, when activity is low.

Eventually I'd like to examine other nRF-based hardware with an eye toward implementing similar improvements.

My first work on this is already underway, to do the zero-risk stuff, like enabling a SoftDevice low power mode, turning off unused peripherals, etc.  I would be doing multiple pull requests, split up by risk level, broadly speaking.

I would appreciate feedback on whether this work would be welcomed, and what conflicts it might have with current development or planned features.

taedryn, 2025-08-25:
Well, certainly the zero-risk stuff like turning off peripherals seems like low-hanging fruit.  This is stuff that's literally not connected to anything on the board.  Leaving it on is just warming the board for no reason.

I was looking at the BT stuff, but given that we already have complaints about nodes being hard to connect with, I'm not inclined to do anything there, at least until I understand the code a bit better.

fdlamotte, 2025-08-26:
I'll be happy to follow your work @taedryn 

There has been some work done on T1000, but not on T114 AFAIK