meshcore-dev/MeshCore #670: WiFi Companion Sometimes Refuses Connections until rebooted

lincomatic, 2025-08-26:
The WiFi companion sometimes will not connect unless the node is rebooted. This happens more often on Android, and happens whenever the phone loses WiFi connectivity. An easy way to reproduce the issue is to turn WiFi on/off on the phone to simulate going out of WiFi range.

I connected a node to a serial console, and saw that the issue is caused by the TCP connection not being closed. On iOS, the WiFi connection opens and closes properly when you WiFi is turned on/off, so this issue is not as prevalent, but it probably will happen if the app crashes.

In SerialWiFiInterface.cpp, there is the following code:

```
if (!client) client = server.available();

  if (client.connected()) {
    if (!deviceConnected) {
      WIFI_DEBUG_PRINTLN("Got connection");
      deviceConnected = true;
    }
  } else {
    if (deviceConnected) {
      deviceConnected = false;
      WIFI_DEBUG_PRINTLN("Disconnected");
    }
  }
```

I debugged the code, and the underlying WiFiClient library still returns client.connected() when the connection is lost in the above described manner, and couldn't think of a way to fix it. I found a simple workaround...

replace

```
if (!client) client = server.available();
```

with

```
WiFiClient newClient = server.available();
  if (newClient) {
    client.stop();
    client = newClient;
    WIFI_DEBUG_PRINTLN("Closing old client");
  }
```

My fix has a side effect that whenever a new client tries to connect to the node, the existing one will get kicked off and replaced by the new one. Personally, I like this behavior, but it deviates from the original.

If the fix is acceptable, I will submit a PR

liamcottle, 2025-08-29:
Looks good to me! Thanks for digging into this :) I've sent through a quick PR, hoping we can get this in for the next release.

liamcottle, 2025-08-30:
Fixed by #685

closed 2025-08-30 by liamcottle