meshcore-dev/MeshCore #681: Auto add contacts options

mikecarper, 2025-08-28:
Have 3 radio buttons for 
Repeaters
Rooms 
Clients

Default is auto add repeaters

focaldistorsion, 2025-09-01:
I'm thinking toggles would work best.  
Let's say you are travelling to a new town.
You wouldn't want all clients but repeaters only repeaters and rooms. 
Or any combo based on needs.

ericszimmermann, 2026-01-16:
I would even prefer a fourth option: Auto Add on incomming Direct Message, if possible. So that not even all users will be added, but only those for those which communication may be established. It would have to be handled not in anouncement but incomming message part, but makes able to receive messages from contacts without having to have added the contact beforehand.
Sincerely Eric

closed 2026-01-16 by liamcottle