meshcore-dev/MeshCore #709: BT Auto reconnect

focaldistorsion, 2025-09-01:
Hey there! 

First of all congratulations on making the MeshCore such a pleasant experience! ⭐⭐⭐⭐⭐

I've been using it constantly as we're rolling out the local mesh and have noticed that when I'm leaving my companion behind for a trip around the house it disconnects - thick concrete walls and node forgotten in a corner. 

When I return in its vicinity it doesn't reconnect. And I've noticed I tend to forget go in the app and manually reconnect. 

It would be a nice to have feature. 😆 Can it do that? 

Or perhaps a an additional one-time notification beeping that it disconnected, aside from the persistent one. 

Running Android 15

hofmand, 2025-09-02:
For me, sometimes it reconnects and other times it doesn't, at least not right away.

focaldistorsion, 2025-10-15:
I am closing this issue for the MeshCore Android. 

App v1.31.0 with FW 1.9.0 seems to have  solved the issue. 

Thanks you @liamcottle 😀👍

closed 2025-10-15 by focaldistorsion