meshcore-dev/MeshCore #733: [Feature Request] Ability to block certain repeater neighbors.

michaeljaddy, 2025-09-04:
Our mesh is receiving interference from a neighbouring mesh - the neighbouring repeater is a high powered node so our repeater can hear it, but it can't hear us so we currently have one way traffic.

Would it be possible to implement something where repeaters would be able to drop traffic from specified repeaters to stop unnecessary flood packets flooding our mesh?

drewzh, 2025-09-08:
I just wanted to add that I know this is affecting a bunch of areas of the UK as there are unscrupulous people adding amplifiers to their repeaters, which is spilling communications over into area where the recipients can never response. I suspect that it's getting very frustrating for them as I know some have resorted to switching off hilltop repeaters so that they don't have any possibility of 'bridging' out until they have a better solution.

drewzh, 2025-09-08:
Also see https://github.com/meshcore-dev/MeshCore/issues/742

MCRDavid, 2025-09-13:
Can I ask if the idea to create a "blacklist" goes ahead, that the project uses a term of block list, deny list, or something similar please?

I appreciate it may be divisive among some, but would ask it's use be reconsidered. 
Some reference articles as to why we should consider dropping the terms white/black list below

https://www.selfdefined.app/definitions/blacklist-whitelist/

https://hodigital.blog.gov.uk/2020/07/23/not-a-black-and-white-issue-using-racially-neutral-terms-in-technology/

https://www.ncsc.gov.uk/blog-post/terminology-its-not-black-and-white

https://www.splunk.com/en_us/blog/learn/blacklist-whitelist-inclusivity.html

closed 2026-09-11 by github-actions[bot]