meshcore-dev/MeshCore #742: Traffic Shaping needed for repeaters

AndyKirby, 2025-09-05:
Some discussion, way too early for feature requests etc. There has arisen a number of instances where some sort of traffic shaping through a repeater/repeater cluster was needed. Is there any planning or thought heading this way ??

Maybe some coarse traffic shaping through Black/Grey/White listing or finer with targeted rate limiting.

fdlamotte:-

I think this has been envisioned from the start, I remember my first reads about scott's vision for repeaters ...
so we should at some point address this

It will be an interesting discussion,  we don’t really want to mute users to isolation but we do need to somehow shape the traffic in favour of inclusion, with out killing message delivery reliability. Any solution will be imperfect. The question though is which one is the least imperfect and fits in the available resources of the average repeater node.
There is of course the option to stream traffic via channels etc, but as all channels effectively flood, doing so does not actually reduce or shape the traffic, just hides it from other repeater users whilst consuming the same resources.

fdlamotte:-

You might start a issue (if not done already) to centralise all that

AndyKirby, 2025-09-05:
BTW A passing thought that happened while submitting the above.

There is a question as to whether applying Black/Grey/White Listing or rate limiting to channels will get more juice for the squeeze than say applying similar to nodes attempting to pass traffic through the repeater.

You basically have to tag or target your rate limiting at something.

The option to shape channels through a repeater would make it possible to keep local SIG's off longer range links for example.

ie channel:- 
City-U-Like-Chat-For-Locals
AreaX-RepeaterGroup

etc etc

drewzh, 2025-09-08:
I suspect this would also apply to https://github.com/meshcore-dev/MeshCore/issues/733