meshcore-dev/MeshCore #746: Companion firmware v1.8.1 keeps crashing on a Heltec T114

hotwolf, 2025-09-06:
I've installed Companion Firmware v1.8.1 on my HeltecT114, but it is not usable, because it freezes up and terminates the Bluetooth connection every 1 or 2 minutes. The same T114 hardware runs fine with companion firmware v1.7.4.
I have no problems running Companion Firmware v1.8.1 on my Heltec V3.

Please let me know is there is any information I can provide to help find the cause  of this issue.

recrof, 2025-09-07:
Hello, do you use iOS phone to connect? can you tell us what version of iOS and model of the phone?

hotwolf, 2025-09-07:
Hello, I'm using Android phones (version 15 and 12) for the Bluetooth connection. 
However the crashes also occur without any Bluetooth connection. After a short while, the T114 freezes (display stays off when pushing the button and it does not receive direct messages anymore) even when no phone is connected.

recrof, 2025-09-07:
this seems like flash corruption, could you please export(backup) your config in app settings menu and then perform factory reset & config import?

hotwolf, 2025-09-07:
That worked for a minute, now the firmware crashes before it leaves the boot screen. This is what I did:

1. Exported my configuration on companion-firmware v1.7.4
2. Did a factory reset 
3. Put the T114 in DFU mode
4. Erased the Flash
5. Flashed companion-firmware v1.8.1
6. Connected the T114 to my phone
7. Did another factory reset
8. Imported the configuration
9. Tried the Bluetooth connection with another phone -> successfull
10. Set a fixed Bluetooth pin (because I'm too slow to type in the random pin before it disappears)
11. Tried the Bluetooth connection with two phones -> successfull
12. Let the T114 run for a while until it stopped working
13. Now it freezes right after the reboot and no Bluetooth connection is possible

mattsains, 2025-09-09:
This is happening to me too with a T1000e on 1.8.1 and iOS 26 beta. I’ll be connected and then it’ll disconnect from the app (which seems to get confused when this happens) and then won’t return until I reset the T1000e. It doesn’t seem to lock up unless I am connected to the device (ie., it functions fine if it’s left alone)

mattsains, 2025-09-09:
That doesn’t help. I’ve tried all of the following:

* force closing the app
* toggle bluetooth
* press/press and hold button on the device

hotwolf, 2025-09-09:
I've made another observation in my case:
I have room server (with repeat option) on my balcony through which my T114 sees up to 24 active repeaters. The connection to all of them seems to be unidirectional. I can receive public messages, but pings to the closest repeaters always fail. When I turn off this room server I don't see any network activity inside my house, but the T114 still crashes as soon as any of my Android phones attempts a Bluetooth connection.

r3wald, 2025-09-16:
Here's what I did:

+ enable UF2 mode on the t114
+ copy Heltec_t114_companion_radio_ble-v1.8.1-NOPOWERSAVE to the t114
+ wait for restart
+ step through all screens by clicking the top button
+ while doing so I trigger an advert on the room server

result is the same as before. the t114 stops working and has to be reset.

BTW: USB works well

oltaco, 2025-09-16:
> Here's what I did:
>
> * enable UF2 mode on the t114
> * copy Heltec_t114_companion_radio_ble-v1.8.1-NOPOWERSAVE to the t114
> * wait for restart
> * step through all screens by clicking the top button
> * [...]

If the advert being received causes the lockup then it suggests filesystem corruption to me. I advise doing a filesystem erase to see if that fixes it.

oltaco, 2025-09-16:
> > Please let us know if this fixes the problem.
> 
> Thanks [@oltaco](https://github.com/oltaco), that seems to have resolved my problem. With the fixed 1.8.1 firmware you've posted, I haven't seen crashes on any of my T114 companions so far.

Thanks for testing, I'll let the other devs know 👍

oltaco, 2025-09-17:
> Here's what I did:
>
> * enable UF2 mode on the t114
> * copy Heltec_t114_companion_radio_ble-v1.8.1-NOPOWERSAVE to the t114
> * wait for restart
> * step through all screens by clicking the top button
> * [...]

Just to make sure, you've been running 1.8.1 from the flasher, and the firmware I posted. Not anything from the dev branch right?

hofmand, 2025-09-26:
Did the power saving code make it into the "RAK Wireless WisBlock / WisMesh (RAK 4631)"? Because I'm seeing a similar issue where if my phone goes out of bluetooth range, the companion node eventually stops receiving messages and the app can't reconnect to it until I press the physical button on the device. The RAK 4631 is a "nRF52840 SX1262" device like the T114. Mine is running v1.8.1.