meshcore-dev/MeshCore #752: Open up SF 5 & 6

mikecarper, 2025-09-08:
Right now SF 7 is as low as we can go. Can we open this up to even lower SF values of 5 & 6? Needs the same PPM requirements ( https://www.semtech.com/design-support/lora-calculator ). SF6 output produced by an SX1262 cannot be read by the SX127x devices (which is heltec v2's only I believe). Could also go all the way down to 20.8khz bandwidth it seems https://stuartsprojects.github.io/2019/10/08/sx1262-improved-LoRa-device.html

recrof, 2025-09-08:
sx1276 devices: v2, tbeam, tlora 2.1_1.6.

moverhorst-pitcrew, 2025-10-22:
A very large portion of the nodes used are SX1262, so probably would stick to compatible settings.

recrof, 2025-10-22:
opened PR: https://github.com/meshcore-dev/MeshCore/pull/993
this needs to be refined more, as sx1276 will error out if you use settings like this. we need to bubble the RadioLib errors up, so user knows it could not be saved/used.