meshcore-dev/MeshCore #817: Feature: Proper Duty Cycle Calculation

marcelverdult, 2025-09-22:
Basic idea: let’s assume 10% duty cycle in Europe calculated rolling over last hour.

1 hour = 3600sec = 3.600.000 ms -> 10% duty cycle = 360.000ms
We now need 2 variables: tx_budget and last_update

So we start with a full tx_budget of 360.000ms at last_update 0

Now we send first message which takes 200ms 
-> reduce tx_budget by 200 ms -> 359.800ms
-> set last_update to now()

Next message 100secs after first one:
-> check difference between last_update and now()  = 100sec
-> during that time we had no TX so we can add 10% of that ( 10 sec / 10000ms ) back to our tx_budget
-> tx_budget = 360.000ms

Now we send 100 messages in a row each 200ms
-> tx_budget reduced by (100x200ms) 20.000ms
-> tx_budget 340.000ms
-> last_update = now()

5sec later next package which is bigger and takes 300ms:
-> time difference last_uodate and now = 5sec -> add 0.5sec/500ms to budget
-> tx_budget 340.500ms
-> reduce budget by the now needed 300ms
-> tx_budget 340.200ms

That way you always have a rolling window defined by your budget and refilled by duty cycle percentage of passed time without TX.

[image]

marcelverdult, 2025-11-19:
could you PLEASE implement this?

Nutzi99, 2025-11-20:
This is a great idea for countries like Germany with a 10% duty cycle. Many repeaters in good locations would benefit from this.