meshcore-dev/MeshCore #834: Solve spreading factor fragmentation - multi channel gateways

tbaumann, 2025-09-25:
How can we solve communities fragmenting into different spreading factors?
There are good reasons to switch to different radio setups. But there is no routing between clusters. And stragglers who didn't switch in the region are left behind.

Perhaps some gateways could bridge between networks.

At first glance, the routing logic has no concept of radio channels. Which leaves mirroring every packet on a different spreading factor. Which frankly sounds horrible. (Especially since it's still happening on the same broader carrier)
I guess mirroring flood packets is out of the question.

But perhaps a router could keep taps on which channel a next hop was seen before, and choose to route it using a different RF setting? (With extra consideration for only changing RF settings if the next hop is only active on a different channel)

I just wanted to see if this idea has any legs.

Based on that, instead of having two radios, such a higher order gateway could use a LoRa Concentrator chip like SX1302/SX1303. That won't work with radiolib, some heavy refactorying required.

recrof, 2025-09-26:
> I remember Radiolib works for sx13xx too.
> 
> Since sx13xx is cheaper now. It is a good idea to increase the mesh capacity.
> 
> https://github.com/jgromes/RadioLib/tree/master/examples%2FSX123x

that's for SX123X. those chips don't even support LoRa

tbaumann, 2025-09-26:
I think the primary challenge is the routing core. Once that understands different 'carriers' all sorts of other technologies could be used. (FSK, 2.4Ghz LoRa, WLAN, Ethernet, ... -- In fact WLAN alredy exists via ESP-Now)

The radiolib part of meshcore is already quite abstract. As a proof of concept, before implementing concentrator code, one could just use two seperate radios.

tbaumann, 2025-09-28:
https://github.com/meshcore-dev/MeshCore/pull/454 interesting