meshcore-dev/MeshCore #867: Nuovo firmware con possibilità di telcomando

iw1qaf, 2025-09-30:
Buongiorno a Voi. Vi chiedo se potete sviluppare un nuovo firmware che preveda, con l’invio di un COMANDO o di un MESSAGGIO di TESTO, di attivare un piccolo relais in modalità On/Off, in modo da comandare, per esempio, un riavvio da remoto di una apparecchiatura. Per esempio abilitando le GPIO di una schedina Tlora esp32 (Input/Output). GRAZIE molte! (Sarebbe una applicazione utilissima sulla rete meshcore).

mrlmuk, 2025-09-30:
Ciao, probabilmente otterrai una risposta migliore se traduci la tua richiesta in inglese con translate.google.com. L'ho fatto io per te.

This is a feature request asking to be able to control things (GPIO pins) remotely over meshcore, translated below.

Good morning. I'm wondering if you can develop a new firmware that, by sending a COMMAND or TEXT MESSAGE, can activate a small relay in On/Off mode, so as to control, for example, a remote reboot of a device. For example, by enabling the GPIOs of a Tlora ESP32 board (Input/Output). THANK YOU very much! (It would be a very useful application on the meshcore network).

iw1qaf, 2025-09-30:
Crazie! Grazie molte! Attendo una gradita risposta. A presto!
Giovanni

mrlmuk, 2025-10-01:
Non l'ho provato, ma una richiesta simile, supporto GPIO per stanze e ripetitori tramite interfaccia di amministrazione - https://github.com/meshcore-dev/MeshCore/issues/772

suggerisce che è già possibile farlo con il ruolo sensore
comando sensori: gpio - https://github.com/meshcore-dev/MeshCore/pull/547

-----

I have not tried this but a similar request, GPIO support for rooms and repeaters via admin cli - https://github.com/meshcore-dev/MeshCore/issues/772

has the suggestion that you can already do this with the sensor role
sensors: gpio command - https://github.com/meshcore-dev/MeshCore/pull/547

iw1qaf, 2025-10-01:
GRAZIE 1000. Ora mi leggo le informazioni. ancora grazie. Giovanni